pymodbus-dev/pymodbus #533: "infinite loop" takes 100% of CPU when using async RTU (serial) with Tornado

Cougar, 2020-09-13:
### Versions

* Python: 3.8.3
* OS: Linux
* Pymodbus: v2.4.0
* Modbus Hardware (if used): IT5888-4

### Pymodbus Specific
* Client: rtu - async (Tornado)

### Description

When using async serial RTU with Tornado, [BaseTornadoSerialClient](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L135) class is in use. In this class, the method [execute()](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L162) defines [callback()](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L170-L185) for serial data reading.

https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L170-L185

The problem with this callback is that it is called (sometimes?) before any data is available and `self.stream.connection.in_waiting` is zero. Until it is zero, the `while True` loop runs with maximum speed taking all possible CPU time of one core. In case there will be no data coming from serial port, it runs until serial timeout reached i.e. usually many seconds.

### Code and Logs

To see what is going on in this loop, I added a following logging:

```diff
         def callback(*args):
             LOGGER.debug("in callback - {}".format(request.transaction_id))
+            count = 0
             while True:
                 waiting = self.stream.connection.in_waiting
                 if waiting:
@@ -182,7 +183,10 @@
                         unit,
                         tid=request.transaction_id
                     )
+                    if count:
+                        LOGGER.info("waiting=%d, count=%d", waiting, count)
                     break
+                count += 1
```

And this is the log when running code:

```
2020-09-13 13:20:22,894 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=11560
2020-09-13 13:20:22,910 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7127
2020-09-13 13:20:22,926 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7644
2020-09-13 13:20:22,942 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7421
2020-09-13 13:20:22,957 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7620
2020-09-13 13:20:22,973 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7839
2020-09-13 13:20:22,989 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7649
2020-09-13 13:20:23,005 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7734
2020-09-13 13:20:23,021 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7763
2020-09-13 13:20:23,037 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7679
2020-09-13 13:20:23,053 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7783
2020-09-13 13:20:23,069 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7562
2020-09-13 13:20:23,085 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=7742
2020-09-13 13:20:23,101 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=6, count=7671
```

There is something else wrong in the serial logic and my actual problem is not in this loop but whatever it is, this loop should be fixed as well. I'll update this ticket with other findings later with possible solution when I'll find it.

Cougar, 2020-09-13:
When looking some more, it seems me that the whole serial handling logic is incorrect and can work only in ideal situations. In case of any error condition or other delay in data (following example), it just stops the whole Modbus processing.

This is how it works right now: [execute()](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L162) sends the frame to the serial port and sets callback for this [`write()`](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L189) call to read reply. Callback for `write()` call signals that all data has been sent to the serial port, not that there is anything to read from serial port.

Using this [callback](https://github.com/riptideio/pymodbus/blob/de3955e46087f5cdcd77184da9a681d7ad20c032/pymodbus/client/asynchronous/tornado/__init__.py#L170), the `while True` loop just waits for incoming data. After wasting a lot of CPU cycles, this data (hopefully) arrives. But if kernel gives serial port data to the user space before all reply has been received, then the next processing just stops because there is no callback set for further data receiving. Framer detects that some data is missing but no part of code calls `read()` and framer again after that. This is how it looks like in log when the cycle in callback finds some bytes in serial buffer and reads them out:

```
2020-09-14 00:01:21,215 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:192 __init__.execute(): send: 0x1 0x3 0x1 0x1 0x0 0x1 0xd4 0x36
2020-09-14 00:01:21,215 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/transaction.py:398 transaction.addTransaction(): Adding transaction 1
2020-09-14 00:01:21,215 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=7, count=1472
2020-09-14 00:01:21,215 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:171 __init__.callback(): in callback - 1
2020-09-14 00:01:21,236 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:177 __init__.callback(): recv: 0x1 0x3 0x2 0x2
2020-09-14 00:01:21,236 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/framer/rtu_framer.py:241 rtu_framer.processIncomingPacket(): Frame - [b'\x01\x03\x02\x02'] not ready
2020-09-14 00:01:21,236 INFO     ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:187 __init__.callback(): waiting=4, count=4418
```

After returning from callback program just hangs. Actually, it sleeps in Tornado IOloop and serial data is waiting in the serial buffer. When user sets its own timer, it is possible to get out from this situation, but unread data in serial buffer creates a big mess. This is how it goes on when there is 10 sec timeout set and next query goes out:

```
2020-09-14 00:01:31,225 DEBUG    ioloop         ./test.py:48 test._send_request(): _send_request(498)
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:192 __init__.execute(): send: 0x1 0x3 0x1 0xf2 0x0 0x1 0x24 0x5
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/transaction.py:398 transaction.addTransaction(): Adding transaction 1
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:171 __init__.callback(): in callback - 1
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/client/asynchronous/tornado/__init__.py:177 __init__.callback(): recv: 0x6b 0xf8 0xcb
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/framer/rtu_framer.py:186 rtu_framer.getFrame(): Getting Frame - 0x3 0x2 0x2 0x6b
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/factory.py:266 factory._helper(): Factory Response[ReadHoldingRegistersResponse: 3]
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/framer/rtu_framer.py:115 rtu_framer.advanceFrame(): Frame advanced, resetting header!!
2020-09-14 00:01:31,227 DEBUG    ioloop         /tmp/testx/.venv/lib64/python3.8/site-packages/pymodbus/transaction.py:409 transaction.getTransaction(): Getting transaction 1
```

Here we can see that the Tornado reads `0x6b 0xf8 0xcb` first, which actually is the end of previous reply. I also checked from Modbus device (with logic analyzer) that this data was sent out just after `0x01 0x03 0x02 0x02` and not 10 sec later..

[image]

Cougar, 2020-09-15:
With "small" patch I got it working as it should in async mode.

I didn't find any existing method that can be used to check partial frames and let data reading continue if frame is not ready. All other implementations are broken one or another way. They either drop buffers if frames are not fully read, especially when last 1-2 bytes (CRC) is missing, or they are not able to process buffers with 1 byte in it like in [ModbusRtuFramer.decode_data()](https://github.com/riptideio/pymodbus/blob/master/pymodbus/framer/rtu_framer.py#L73) where at least two bytes must always exist but there is no check for that.

I didn't find any clear code to calculate excepted frame size either. There is [BaseTornadoSerialClient.execute()](https://github.com/riptideio/pymodbus/blob/master/pymodbus/transaction.py#L261) where such calculation is used but it is quite a big spaghetti and should be moved out to separate (public) function(s) for general use. I took some lines of code from that but to keep my code short I used local methods from other classes which is not nice.

I also added recv buffer cleanup code from sync [ModbusSerialClient._send()](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/sync.py#L643) to make sure that there is no leftover in serial port buffer when sending new request. This is not the best solution as it takes time in every transaction but the problem itself is very rare. It would be better to do this cleanup only in case of error conditions and timeouts. This check takes 1.4 ms at the moment.

I still don't check if the response is Modbus exception. For that, it needs to be checked if response size is equal with Exception frame size and if it is an Exception. Right now I don't do any timeout checking either. But otherwise it works well.

I added two callbacks where first callback for write sets the second callback for read. When all data has been received then read callback gets removed. Ideally it would be better to set up read callback only once during the serial port initialization to be always ready to read data. But then some more care should be taken to match incoming data with sent requests because the send and receive parts are not connected with each other in any way. I'm not fully aware how all this transaction logic works. Also timing should be monitored in this case.

I did something similar thing 5 years ago when I implemented [async RTU](https://github.com/droid4control/pymodbus_async/blob/master/client.py#L201) with on top of pymodbus with [silent interval](https://github.com/droid4control/pymodbus_async/blob/master/client.py#L131) check. This code still runs in production very well.

With my test program and controller (19200 baud) I read one HoldingRegister and expect response of 7 bytes. Most times the first call of the read callback returns only 4-6 bytes and then it gets another run where remaining bytes are read. I rarely see all 7 bytes ready for one read and I have not seen more than two callback calls yet.

```
2020-09-16 02:32:37,850 DEBUG    ioloop         ./test.py:49 test._send_request(): _send_request(149)
2020-09-16 02:32:37,850 INFO     ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:227 __init__.execute(): expected_response_length = 7
2020-09-16 02:32:37,850 DEBUG    ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:229 __init__.execute(): send: 0x1 0x3 0x0 0x95 0x0 0x1 0x94 0x26
2020-09-16 02:32:37,850 DEBUG    ioloop         /tmp/testx/pymodbus/transaction.py:398 transaction.addTransaction(): Adding transaction 1
2020-09-16 02:32:37,865 INFO     ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:198 __init__._on_receive(): waiting = 5
2020-09-16 02:32:37,865 DEBUG    ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:200 __init__._on_receive(): recv: 0x1 0x3 0x2 0x0 0x0
2020-09-16 02:32:37,881 INFO     ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:198 __init__._on_receive(): waiting = 2
2020-09-16 02:32:37,881 DEBUG    ioloop         /tmp/testx/pymodbus/client/asynchronous/tornado/__init__.py:200 __init__._on_receive(): recv: 0xb8 0x44
2020-09-16 02:32:37,881 DEBUG    ioloop         /tmp/testx/pymodbus/framer/rtu_framer.py:186 rtu_framer.getFrame(): Getting Frame - 0x3 0x2 0x0 0x0
2020-09-16 02:32:37,881 DEBUG    ioloop         /tmp/testx/pymodbus/factory.py:266 factory._helper(): Factory Response[ReadHoldingRegistersResponse: 3]
2020-09-16 02:32:37,881 DEBUG    ioloop         /tmp/testx/pymodbus/framer/rtu_framer.py:115 rtu_framer.advanceFrame(): Frame advanced, resetting header!!
2020-09-16 02:32:37,881 DEBUG    ioloop         /tmp/testx/pymodbus/transaction.py:409 transaction.getTransaction(): Getting transaction 1
2020-09-16 02:32:37,881 DEBUG    ioloop         ./test.py:79 test._on_done(): _on_done(149)
2020-09-16 02:32:37,881 INFO     ioloop         ./test.py:100 test._print(): Printing 149: -- [0]
serial  : 149 = [0]
```

It works quite fast, one request+response cycle takes 12.9 ms and delay between these cycles are about 3.45 ms.

This is my current "MVP" patch:

```diff
diff --git a/pymodbus/client/asynchronous/tornado/__init__.py b/pymodbus/client/asynchronous/tornado/__init__.py
index 2b0cc55..17729bd 100644
--- a/pymodbus/client/asynchronous/tornado/__init__.py
+++ b/pymodbus/client/asynchronous/tornado/__init__.py
@@ -184,9 +184,39 @@ class BaseTornadoSerialClient(AsyncModbusSerialClientMixin):
[... cut ...]
+                " ".join([hex(byte2int(x)) for x in result]))
+
         packet = self.framer.buildPacket(request)
+        response_pdu_size = request.get_response_pdu_size()
+        expected_response_length = self.transaction._calculate_response_length(response_pdu_size)
+
         LOGGER.debug("send: " + " ".join([hex(byte2int(x)) for x in packet]))
-        self.stream.write(packet, callback=callback)
+        self.stream.write(packet, callback=_on_write_done)
         f = self._build_response(request.transaction_id)
         return f
 
```

Cougar, 2020-09-23:
I implemented a "usable" asynchronous serial/RTU client library on top of `pymodbus` in [https://github.com/Cougar/pymodbus_async](https://github.com/Cougar/pymodbus_async).

This is what I did:
  * Replaced (extended) [`pymodbus.client.asynchronous.serial.AsyncModbusSerialClient`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/serial.py#L16) to not use original [`pymodbus.client.asynchronous.factory.serial.io_loop_factory()`](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/asynchronous/factory/serial.py#L59) but a new [`pymodbus_async.client.asynchronous.serial.async_io_loop_factory()`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/serial.py#L6) to use a new [`pymodbus_async.client.asynchronous.tornado.AsyncModbusSerialClient`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L147).
  * Added new exception [`TimeOutException`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/exceptions.py#L3) to notify application code about serial reading timeout. With this new exception, all possible events are covered and caller __always__ gets some answer: either a normal Modbus reply, Modbus exception or exception related with serial I/O including timeout. With that it is safe in application code to send Modbus command and continue flow in callback instead of having some additional internal timeout mechanism . This is important because with timeout handling in application code, it is impossible to clean up possible hanging states in `pymodbus` lib. For instance, the transaction response [Future](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/asynchronous/tornado/__init__.py#L114) may remain awaiting infinitely and therefore leaks memory.
  * Extended [`pymodbus/client/asynchronous/tornado/__init__.py`](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/asynchronous/tornado/__init__.py) where main code is. This is a complete rewrite of [`AsyncModbusSerialClient.execute()`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L23) to handle I/O events and be totally asynchronous. Logic is the same that I described earlier i.e. this code [clears](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L128-L136) serial buffer from any leftover data, makes sure that there is at least 3.5 char long [delay](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L138-L143) between commands sent to Modbus, that reply reading starts just after [successful command sending](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L65) and reply reading [timeout](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L104) is [handled](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L37-L59).

I'm not sure if the [`close()`](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L79) alone is enough when waiting for serial data but I/O exception (disconnect for instance) occurs or is there anything else that should be done? The similar code is probably needed in other places too, where serial port has been read (to clear buffer) or [written](https://github.com/Cougar/pymodbus_async/blob/master/pymodbus_async/client/asynchronous/tornado/__init__.py#L145)?
 
Related thing that is still not fully handled is I/O error that comes from disconnecting USB serial port when code is in Tornado IOLoop and which raises an exception. I don't know how to gracefully handle that. For me it looks like this is more an application responsibility and even can't be handled in `pymodbus` but correct me if I'm wrong. Anyway, awaiting Future needs to be finished in this case as well (is current timeout handler enough for that?) This needs more testing but as it is not a normal Modbus related event, I let it this way right now..

Maybe it would be good idea to add a new (private) method for all kind of serial port reading (normal reply reading and buffer cleaning) to avoid [DRY](https://en.wikipedia.org/wiki/Don%27t_repeat_yourself). Or even use [`eset_input_buffer()`](https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.reset_input_buffer) instead. It is probably much faster but removes possibility to debug print buffer contents.

Also, I'd like to have [response length calculation](https://github.com/riptideio/pymodbus/blob/master/pymodbus/transaction.py#L90) method available outside `ModbusTransactionManager` class. In my code, I just use this _private_ method, which is not correct way to do that. 

@bhagn or @dhoomakethu, please review a code in my [repo](https://github.com/Cougar/pymodbus_async/tree/master/pymodbus_async) and comment/criticize my ideas. If that looks fine in general, then I'd like to submit a PR to finalize this implementation and include it in `pymodbus`.

dhoomakethu, 2020-09-24:
@Cougar thanks for the analysis. I will take a look. In general, I am planning to keep only asyncio support from 3.x series for async client and servers as its real pain to maintain 3 async frameworks (twisted, tornado and asyncio). Also, I am planning to completly redo many parts for 3.x (some you have mentioned in your comments).

dhoomakethu, 2020-09-24:
@Cougar What version of tornado are you supporting (6.x?)

Cougar, 2020-09-24:
> @Cougar What version of tornado are you supporting (6.x?)

I used the version that you suggested earlier, i.e. my module works with `tornado==4.5.2` and `pymodbus==2.4.0`

dhoomakethu, 2020-09-30:
@Cougar this looks good, would you mind raising a PR please?

dhoomakethu, 2020-12-30:
Fixed with 2.5.0.
Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
Refer https://github.com/riptideio/pymodbus/releases

Cougar, 2021-01-01:
> Fixed with 2.5.0.
> Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
> Refer https://github.com/riptideio/pymodbus/releases

Looks good. Still would be great to do some code cleanup and refactoring for 3.0 but it is usable now.

I let my [test](https://github.com/Cougar/pymodbus_test/tree/pymodbus-2.5.0rc2) run more than 7 hours (443 min) at 19200 doing more than 900k registry read transactions. It took only 15 min user CPU time and less than a minute system CPU time.

I also tested it with two serial instances running in parallel and that also worked fine.

closed 2021-03-03 by dhoomakethu