pymodbus-dev/pymodbus #538: Previous response from modbus device returned as current response when using sync TCP over network with lag

thormick, 2020-10-06:
### Versions

* Python: 3.8.2
* OS: Ubuntu 20.04.1 LTS
* Pymodbus: 2.4.0
* Modbus Hardware (if used): Hardware independent, verified with local testing against dummy sync server

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description
TCP sync requests over a laggy network connection against different ranges of registers eventually give the timed-out result of one register range performed previously as the reading of the other register range.

Because the transaction manager and socket framer ignore the transaction ID of the response (my client's app uses synchronized TCP requests) and just returns the response that has come in last, pymodbus can under certain circumstances silently return the result of the previous request made as if it was the result to the one being currently made. This is Bad for my client, whose application makes sequential modbus requests to different ranges of registers of a single unit and needs to communicate over networks with sporadic periods of high lag (problematic VPN connections, mobile networks, etc...).

A step-by-step description of what happens, where an application is set up to read two or more separate ranges of registers in sequential fashion against a single device, and there is network lag. What will happen sooner or later is:

1. Read against register range 1 fails due to timeout. `ModbusTransactionManager._recv` raises `InvalidMessageReceivedException`. This is caught in `ModbusTransactionManager._transact`, which closes the connection, and as such the data will not cause an issue when it *does* arrive. `ModbusTransactionManager.execute` registers this as that the device didn't respond in a timely fashion, and adds it to the `_no_response_devices` list. My client's application sees that there is no valid response, logs an error and sleeps a few seconds.
2. Read against register range 1. pymodbus sees the connection is closed and opens it. There is another timeout, but because the device this time is already on the `_no_response_devices`, there is no exception from `ModbusTransactionManager._recv`, even if it fails to read any data before the timeout, and the connection is *not* closed. `ModbusTransactionManager.execute` just returns an exception with "No Response received from the remote unit/Unable to decode response". My client's application sees that there is no valid response from this register range, logs an error and sleeps a few seconds.
3. Read against register range 1. This time the response to this request, again, does not arrive before the timeout. But: The connection has been left open and the response from previous request does arrive in time (or arrived while the application was sleeping for some seconds, more likely). The transaction ID is one less than it should be, but this is ignored by pymodbus, and it's a result from the right range of registers, just delayed whatever duration the main loop is configured to sleep for. `ModbusTransactionManager.execute` removes the device from the `_no_response_devices` list. My client's application parses this as normal and stores it to processes it and pass it on when all reads are done.
4. Read against register range 2 is attempted. Again there is a delay and the response to this does not arrive before the timeout, but the previous one has already arrived. But this is a response to the previous request against the other register range, and still it's just parsed and processed as normal data from register range 2. The TID is still one less than it should be and it might even be longer than expected, but all of this is ignored. Hilarity ensures, ints are interpreted as floats and the wrong scaling is applied by my client's application and inserted into their database, chaos propagates down the line, alarm messages are sent indicating that more power has been drawn from an outhouse light bulb in fifteen minutes than what the Eurasian continent normally uses in a year, my client's client's boss calls my client's boss and complains that there is junk in reports, human sacrifice, cats and dogs living together, etc, etc...
5. This keeps on for as long as the network lag is over the timeout threshold. Once it falls below that two responses arrive at the same time, the first of which is ignored (addTransaction is invoked with the same TID twice, overwriting the first), and so everything seems to magically fix itself.

The case where a late-returned response is shorter than expected is also Not Good, but in this case the application will hit a "IndexError: list index out of range" before it passes on junk.

### Code and Logs
To test this I I've used a local dummy server based on [synchronous server](https://pymodbus.readthedocs.io/en/latest/source/example/synchronous_server.html) on port 5020, that just returns constants:
```python
    values = [1, 0, 2, 0, 3, 0, 4, 0 ...etc, etc... 63, 0, 64, 0]
    msdb = ModbusSequentialDataBlock(0x00, values)
```

`msdb` is then used for each initializer of the `ModbusSlaveContext`.

Then I add lag to port 5020:

```bash
sudo tc qdisc add dev lo root handle 1: prio priomap 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
sudo tc qdisc add dev lo parent 1:2 handle 20: netem delay 550ms 500ms 25%
sudo tc filter add dev lo parent 1:0 protocol ip u32 match ip sport 5020 0xffff flowid 1:2
```

To highlight the issue I've added the following lines to `DictTransactionManager.addTransaction` just before the last line:
```python
        if tid != request.transaction_id:
            _logger.warning("Adding transaction with incorrect TID!!!! "
                            f"Using TID {tid}, response TID: {request.transaction_id}")
        if tid in self.transactions:
            _logger.warning(f"Replacing transaction {tid}")
```

Then I configure the client's application with `timeout=0.75`, to output to console instead of a database, and use one of their configurations that does three register ranges against a single modbus device. I start the application, which creates a modbus client and tries to do three `read_holding_registers` calls against different register ranges in a loop. Eventually this causes one of the ranges to be returned as another range.

A log of a run like this, edited to remove most logging by my client's application, (mostly debug-type messages that they use to make sure they've got scaling set right on their side and such). The first run is successful, the next two fail, and the last one returns junk. My client uses the term "group" to describe preconfigured ranges of registers:

```
2020-10-06 14:03:59,325 - DEBUG - pymodbus.transaction - Current transaction state - TRANSACTION_COMPLETE
2020-10-06 14:03:59,326 - DEBUG - pymodbus.transaction - Running transaction 9
2020-10-06 14:03:59,326 - DEBUG - pymodbus.transaction - SEND: 0x0 0x9 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x28
2020-10-06 14:03:59,326 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-06 14:04:00,078 - DEBUG - pymodbus.transaction - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-10-06 14:04:00,078 - DEBUG - pymodbus.transaction - RECV: 0x0 0x9 0x0 0x0 0x0 0x53 0x1 0x3 0x50 0x0 0x0 0x0 0x2 0x0 0x0 0x0 0x3 0x0 0x0 0x0 0x4 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x0 0x0 0x7 0x0 0x0 0x0 0x8 0x0 0x0 0x0 0x9 0x0 0x0 0x0 0xa 0x0 0x0 0x0 0xb 0x0 0x0 0x0 0xc 0x0 0x0 0x0 0xd 0x0 0x0 0x0 0xe 0x0 0x0 0x0 0xf 0x0 0x0 0x0 0x10 0x0 0x0 0x0 0x11 0x0 0x0 0x0 0x12 0x0 0x0 0x0 0x13 0x0 0x0 0x0 0x14 0x0 0x0 0x0 0x15
2020-10-06 14:04:00,078 - DEBUG - pymodbus.transaction - Adding transaction 9
2020-10-06 14:04:00,079 - DEBUG - pymodbus.transaction - Getting transaction 9
2020-10-06 14:04:00,079 - DEBUG - pymodbus.transaction - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-10-06 14:04:00,079 - INFO - util.ms_input - Read register with values [0, 2, 0, 3, 0, 4, 0, 5, 0, 6, 0, 7, 0, 8, 0, 9, 0, 10, 0, 11, 0, 12, 0, 13, 0, 14, 0, 15, 0, 16, 0, 17, 0, 18, 0, 19, 0, 20, 0, 21]
2020-10-06 14:04:00,086 - DEBUG - pymodbus.transaction - Current transaction state - TRANSACTION_COMPLETE
2020-10-06 14:04:00,086 - DEBUG - pymodbus.transaction - Running transaction 10
2020-10-06 14:04:00,086 - DEBUG - pymodbus.transaction - SEND: 0x0 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x28 0x0 0x28
[... 326 lines cut ...]

* Change the last bit of ModbusSocketFramer._process to:
  ```python
            callback(result, tid=result.transaction_id)
  ```
[... cut ...]
* Always close the connection if there is no response at all before timeout, not just only if the device is not on the `_no_response_devices` list.

* Log something when a transaction response is overwritten in `DictTransactionManager.addTransaction`.

On my side I've added a check that the returned response has the right size in the client, but it would be nice if I could rely on pymodbus not returning responses with the wrong length. I'd add a check on the transaction ID there but that's not exposed.

I'm also discussing a switch to using async TCP with my client.

thormick, 2020-10-12:
Quoting myself:
> I'd add a check on the transaction ID there but that's not exposed.

Even though the TID used for the request isn't returned as part of the response object, I noted that the ModbusTcpClient of course has transaction with the last TID used there, that I can check against. The client's app is a quite single threaded and synchronous affair, so there shouldn't be any issue with race conditions.

My client's code now checks the response to `read_holding_registers`/`read_input_registers` this way:

```python
            if response.transaction_id != modbusTcpClient.transaction.tid:
                raise Exception(f"pymodbus returned a response with transaction ID {response.transaction_id} but the "
                                f"highest transaction manager transaction ID is "
                                f"{modbusTcpClient.transaction.tid}. This could happen because of a pymodbus "
                                "bug triggered by network lag; increase timeout.")
            elif len(response.registers) != register_length:
                raise Exception(f"pymodbus returned a response of {len(response.registers)} word length "
                                f"to a request of {register_length} words. Values: {response.registers}")
```

Local testing suggests that this resolves the issue, but I've yet to try it against actual modbus hardware in production.

mahadi, 2020-10-16:
I am observing similar problems. Checking the transaction IDs gives me:

```
AssertionError('
    pymodbus returned a response with transaction ID 16292 but the highest transaction manager 
    transaction ID is 1. 
    This could happen because of a pymodbus bug triggered by network lag; increase timeout.
)
```

thormick, 2020-10-26:
Had a few minutes to peek at this today. I can reproduce this in 1.5.0 8f3fc71035ef0a23dca1214cef67bdd67d5ff83e, but not in 1.4.0 4ed1f1a9750bc931c100ed24f5cc0fc032ab5f4d

thormick, 2020-10-27:
I've toyed around with minimal fix to this in [this repository](https://github.com/gapitio/pymodbus/tree/gapit-fixes-1) (along with some other changes), but it's tricky to do a proper fix. I've gotten the "Always close the connection..." alternative I've outlined done in [this commit](https://github.com/gapitio/pymodbus/commit/8c5ab8f7d435515c97c7545d31b577bbc9391bed), and it seems to improve the issue. But I've only tested this with a local dummy server, and have no idea yet if it resolves the issue sufficiently with real hardware on a real problematic network.

mahadi, 2020-10-27:
You are writing that the transaction management currently is a mess. So this sounds like a redesign is needed?

dhoomakethu, 2020-10-28:
@thormick thanks for the effort, I will take a look when I get some time.

thormick, 2020-10-29:
@mahadi Well not a *full* redesign, the basic design seems sound. There just seems to be a lot of code for handling special cases that is just jammed sideways into the `execute`, `_transact` and `_recv` methods, which could be beneficial to refactor. This makes it difficult to cut through it to address a specific issue.

There's a few other issues that clutter the source, for instance I question the validity of having `isinstance` if-trees instead of having the framers provide this information. I'm aware it's a legitimate style in special cases, but if moving it into the framers is a better approach I'd be happy to refactor this too.

That said, I'm still looking at this so I might come up with a non-invasive and more complete fix.

thormick, 2020-11-06:
I've tweaked my patch a bit, and it shouldn't close the socket more than reasonable now. Available [here](https://github.com/gapitio/pymodbus/tree/gapit-fixes-2) (note that branch contains all the tweaks I've done for [Gapit](https://github.com/gapitio)). I haven't verified that there aren't still weird corner case issues if there is packet fragmentation and only the first part the message arrives before the timeout.

As an aside, AFAICT the dict transaction manager isn't actually used as advertised by the sync clients. With how the it only uses the current TID that it itself generates, and the socket framer doesn't use the response TID, the DTM only uses one slot and then reads and empties that slot. So it just becomes an awkward way to pass the response from one part of the execute method and a couple of lines down. The sync part of the transaction manager is over-engineered anyhow, but it's tricky to see how it could be improved without moving things into the framers, which is also used by the async code too and as such needs to be handled with care.

Either way, without a more complete check that the response is what the caller wants this is still very fragile and sensitive to future changes in the code, so I would recommend that one just avoids the sync clients in favor of the async, even with my fixes.

And to clarify a few points regarding this for people looking at this and wondering whether it applies to them. This bug report and my fix *only* pertains when:
 * Using synchronized TCP (Might affect synchronized UDP, fix might improve the condition but shouldn't fix it; I've only tested ModbusTcpClient anyhow)
 * There are first at least two failed attempts to do a read from a modbus unit
 * One gets a response which is a correct response to one of those earlier failed read attempts
 * The issue goes away if the timeout is set high enough

If not all of this is the case, the issue you're facing is something else than what this bug report and fix addresses.

dhoomakethu, 2020-11-06:
Refer #560 . This is one of the identified bug and a fix is under testing.

thormick, 2020-11-06:
Oops, gave the wrong link to the repo, should be [here](https://github.com/gapitio/pymodbus-1/tree/gapit-fixes-2). Updated my previous message.

mabl, 2022-01-10:
I still observe this issue on TCP sync client if transmission is interrupted by `KeyboardInterrupt`. I think this is caused by the reconnect code not catching this exception. Maybe consider making try catch clause more indiscriminate?

janiversen, 2022-08-27:
Previous response are no longer returned on current dev.

closed 2022-08-27 by janiversen