pymodbus-dev/pymodbus #544: Drop Python 2 support?

swamper123, 2020-10-15:
### Versions
* Python: >=3.6
* Pymodbus: >= 2.4

### Description

I haven't found in the Readme, which Python versions shall be supported  here ("it should work fine under any python version > 2.7 (including python 3+)").

Python 2 is dead since this year (and may have been dead a long time ago, I mean the lifecycle was more than 9 years), so there won't be more fixes, it will just cause more problems to be compatible with future Python versions.
There been have many things happening in the Python 3 world as well, esepcially in the async section with Python3.6.

Being compatible across these worlds makes the code more complex, slower and unsafer, especially if the Python3 developement goes on.

I would recommened to drop support for all Python versions <3.6 (or even start with 3.7 and go up to 3.9) and explicit mention that in the README.

dhoomakethu, 2020-10-28:
Python2 support will be removed from pymodbus 3.x onwards. JFYI.

swamper123, 2020-10-28:
Thanks for the information.
I will start some cleanup PRs already ;)

swamper123, 2022-03-31:
With the new default branch 3.0.0 and #564 merged, this issue is solved.
Case closed. :detective:

closed 2022-03-31 by swamper123