pymodbus-dev/pymodbus #547: Extended address support

SEC-VT, 2020-10-16:
### Versions

* Python: 3.7
* OS: Debian
* Pymodbus: 2.4.0
* Modbus Hardware (if used): 
KOYO CLICK C0-11ARE-D

### Pymodbus Specific
* Client: tcp

### Description

Modbus TCP client is unable to accept addresses > 65535 as input, which limits access to devices using 6 digit addressing.

### Code and Logs

```python
from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('192.168.1.146', port=502)
result = client.read_coils(100033,1)
print(result.bits[0])
client.close()
```

> Traceback (most recent call last):
>   File "modtest.py", line 4, in <module>
>     result = client.read_coils(100033,1)
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/client/common.py", line 48, in read_coils
>     return self.execute(request)
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/client/sync.py", line 108, in execute
>     return self.transaction.execute(request)
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/transaction.py", line 158, in execute
>     broadcast=broadcast
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/transaction.py", line 228, in _transact
>     packet = self.client.framer.buildPacket(packet)
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/framer/socket_framer.py", line 206, in buildPacket
>     data = message.encode()
>   File "/home/user/.local/lib/python3.7/site-packages/pymodbus/bit_read_message.py", line 34, in encode
>     return struct.pack('>HH', self.address, self.count)
> **struct.error: 'H' format requires 0 <= number <= 65535**

tracernz, 2020-10-17:
Are you trying to read a digital input (1xxxxx address, not possible) or a coil (0xxxxx)? 65535 is the highest possible address for the Modbus protocol. [This Kepware page might help](https://www.kepware.com/en-us/support/knowledge-base/2008/modbus-5-digit-addressing-versus-6-digit-addressin/).

SEC-VT, 2020-10-22:
It seems to me that it is indeed possible to read input bits/coils at addresses like 1xxxxx so long as one strips the leading digit. For the above case where I'm attempting to read an input point with MODBUS 984 address 100033:

```python
from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('192.168.1.146', port=502)
address = 33
result = client.read_discrete_inputs(address)
print(result.bits[address-1])
client.close()
```
provides the correct result.

IMO, ignoring the primary table identifier (or rather, implying it in the function call) seems a little unwieldy, but I get the argument that it's extraneous information.

This brings me to a different issue, which is perhaps just me misusing the module.
Note the line where I print the result:
```python
print(result.bits[address-1])
```
Both read_coils and read_discrete_inputs seem to read every bit starting from 0 up to the address. I've confirmed in Wireshark that the modbus packet is in fact asking for 33 bits from reference number 0. The console client performs the same. Am I doing something wrong or is this the intended behavior?

SEC-VT, 2020-10-22:
Nevermind. I had been using a modified version of the module. Closing this issue.

closed 2020-10-22 by SEC-VT

tracernz, 2020-10-22:
> It seems to me that it is indeed possible to read input bits/coils at addresses like 1xxxxx

Ah, sorry, I had this confused with something else in my head. You can't read discrete inputs on modern Modicon PLCs because they use topological addressing rather than the classic 1xxxx addressing.

> so long as one strips the leading digit.

You always do that for Modbus, e.g. for holding registers you drop the leading 4. There is also an offset of 1 in many devices.