pymodbus-dev/pymodbus #550: How can I read the unit id of the slave from the server

l337quez, 2020-10-20:
### Versions

* Python:
* OS:  Linux
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp//async   --> Python
* Client: tcp//async   --> C++

### Description

 have problems, I have been reading but there are things that I do not understand well. My client is written in C ++, it is a microcontroller, I want to read the unit id of that client from a raspberry pi. Currently the two devices communicate. I have been reading on the internet to be able to print the client's unit id by terminal.

I put as unit id the client the number 1

slaves = {
    0x01: ModbusSlaveContext (co = ModbusSequentialDataBlock (0, [0] * 100), hr = ModbusSequentialDataBlock (0, [0] * 100)),
    0x08: ModbusSlaveContext (co = ModbusSequentialDataBlock (0, [0] * 100), hr = ModbusSequentialDataBlock (0, [0] * 100)),
    }
    context = ModbusServerContext (slaves = slaves, single = False)

so i did this but it gives me error:
pymodbus.exceptions.NoSuchSlaveException: Modbus Error: [No Such Slave] slave - 0 does not exist, or is out of range.

I must read both tcp and RUT clients with modbus, each one will have a number, I want to work with exceptions .. I am working with minimal modbus for rtu, but for tcp I have not been able to read the unit id from the server. All unit id's are known .. for example if there is a tcp slave with the number 8, then it will use a TCP tip routine. Because there will be a sampling time and I must read 200 clients.

### Code and Logs

```python
#!/usr/bin/env python
"""
Pymodbus Server With Updating Thread
--------------------------------------------------------------------------

This is an example of having a background thread updating the
context while the server is operating. This can also be done with
a python thread::

    from threading import Thread

    thread = Thread(target=updating_writer, args=(context,))
    thread.start()
"""
# --------------------------------------------------------------------------- #
# import the modbus libraries we need
# --------------------------------------------------------------------------- #
from pymodbus.server.asynchronous import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer

# --------------------------------------------------------------------------- # 
# import the payload builder
# --------------------------------------------------------------------------- # 

from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.payload import BinaryPayloadBuilder

from pymodbus.framer import ModbusFramer, SOCKET_FRAME_HEADER
# --------------------------------------------------------------------------- #
# import the twisted libraries we need
# --------------------------------------------------------------------------- #
from twisted.internet.task import LoopingCall

# --------------------------------------------------------------------------- #
# configure the service logging   MUESTRA SI HAY UN CLIENTE CONECTADO
# --------------------------------------------------------------------------- #
import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig()
log = logging.getLogger()
# Imprimir el log
#log.setLevel(logging.DEBUG) #hace un log de todo lo que llega y envia

# --------------------------------------------------------------------------- #
# define your callback process
# --------------------------------------------------------------------------- #

def updating_writer(a):
    """ A worker process that runs every so often and
    updates live values of the context. It should be noted
    that there is a race condition for the update.

    :param arguments: The input arguments to the call
    """
    # count= numero de valores a recuperar
    # address= Direccion de inicio, deberia ser cero en hex 0x00
    #register = numero de registros
    log.debug("updating the context")
    address = 0x0
    context = a[0]
    register = 16
    slave_id = 0

    #slave_id = 0x00
    values = context[slave_id].getValues(register, address, count=16)
    print(values) # Data recibida
    print(values[0])
    
    if values[0] == 4 :
        #values[0]=4 # WAITLOGIN 
        values[0]=3 # ACEPTED

        print("4 Revisando  datos del log y cambiar el FLAG")

    if values[0] == 8 :
        print("8 Revisando  datos del log y cambiar el FLAG")

    #values[0]=88 # enviamos o en la bobina 1
    #values = [v + 1 for v in values]
    #log.debug("new values: " + str(values))
    context[slave_id].setValues(register, address, values)

    #data = processIncomingPacket(data, self._handleResponse)
    # data = self.getRawFrame() if error else self.getFrame()
    #framer.processIncomingPacket(data, self._handleResponse)
    #print(data)

def run_updating_server():
    # ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- # 
    # di = entradas discretas. Entradas digitales
    # co = coils.  Salidas digitales (coils).
    # hr = holding registros. Salidas analógicas (holding registers).
    # ir = input registros. Entradas analógicas (input registers).
    
    #store = ModbusSlaveContext(
    #di=ModbusSequentialDataBlock(0, [17]*100),
    #co=ModbusSequentialDataBlock(0, [17]*100),
    #hr=ModbusSequentialDataBlock(0, [4,222,333,444,555,6,7,8,9,10,11,12,13,14,15,16]*100),
    #hr=ModbusSequentialDataBlock(0, [0]*100))
    #ir=ModbusSequentialDataBlock(0, [17]*100))
 
    slaves = {
    0x01: ModbusSlaveContext (co=ModbusSequentialDataBlock(0, [0]*100), hr=ModbusSequentialDataBlock(0, [0]*100)),
    0x08: ModbusSlaveContext (co=ModbusSequentialDataBlock(0, [0]*100), hr=ModbusSequentialDataBlock(0, [0]*100)),
    }

    context = ModbusServerContext(slaves=slaves, single=False) # Single false para multiples esclavos
    #context = ModbusServerContext(slaves=store, single=True) # Single false para multiples esclavos

   
    '''  block = ModbusSparseDataBlock({0x00: 0, 0x05: 1})
    block = ModbusSequentialDataBlock(0x00, [0]*5)

    store = ModbusSlaveContext(di = ModbusSequentialDataBlock.create())
    store = ModbusSlaveContext()

    block = ModbusSequentialDataBlock(0x00, [0]*0xff)
    store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block) '''

    
    # ----------------------------------------------------------------------- # 
    # initialize the server information
    # ----------------------------------------------------------------------- # 
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'pymodbus Server'
    identity.ModelName = 'pymodbus Server'
    identity.MajorMinorRevision = '2.3.0'
    
    # ----------------------------------------------------------------------- # 
    # run the server you want
    # ----------------------------------------------------------------------- # 
    time = 5  # 5 seconds delay
    loop = LoopingCall(f=updating_writer, a=(context,))

    loop.start(time, now=False) # initially delay by time
    StartTcpServer(context, identity=identity, address=("0.0.0.0", 5020))

if __name__ == "__main__":
    run_updating_server()
# LEER TEORIA DE MODBUS AQUI https://koldopina.com/python-modbus/

```

### LOGS

```Unhandled error in Deferred:

Traceback (most recent call last):
  File "/home/ronal/.local/lib/python3.8/site-packages/twisted/internet/base.py", line 1283, in run
    self.mainLoop()
  File "/home/ronal/.local/lib/python3.8/site-packages/twisted/internet/base.py", line 1292, in mainLoop
    self.runUntilCurrent()
  File "/home/ronal/.local/lib/python3.8/site-packages/twisted/internet/base.py", line 913, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/home/ronal/.local/lib/python3.8/site-packages/twisted/internet/task.py", line 239, in __call__
    d = defer.maybeDeferred(self.f, *self.a, **self.kw)
--- <exception caught here> ---
  File "/home/ronal/.local/lib/python3.8/site-packages/twisted/internet/defer.py", line 151, in maybeDeferred
    result = f(*args, **kw)
  File "tcp_mills_test1.py", line 71, in updating_writer
    values = context[slave_id].getValues(register, address, count=16)
  File "/home/ronal/.local/lib/python3.8/site-packages/pymodbus/datastore/context.py", line 179, in __getitem__
    raise NoSuchSlaveException("slave - {} does not exist, "
pymodbus.exceptions.NoSuchSlaveException: Modbus Error: [No Such Slave] slave - 0 does not exist, or is out of range ```

l337quez, 2020-10-21:
i need decode framer unit

dhoomakethu, 2020-10-28:
How are you writing to the server ? I see you have two slaves `0x01` and `0x08` but in `updating_writer` function you are trying to get context with slave_id=0 `values = context[slave_id].getValues(register, address, count=16)`. Try with either of `0x01` or `0x08` and your code should work as expected.

closed 2020-10-28 by dhoomakethu

l337quez, 2020-10-28:
my question and the support plant it wrong. so I'll do another issue, thanks for answering, but now if I'm going to be clearer in my question