pymodbus-dev/pymodbus #551: retry_on_empty broken in 2.4.0

thormick, 2020-10-26:
### Versions

* Python: 3.8.5
* OS: Ubuntu 20.04.1 LTS
* Pymodbus: 2.4.0
* Modbus Hardware (if used): Hardware independent, verified with local testing against dummy sync server

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp - sync

### Description

retry_on_empty feature doesn't work without also setting retry_on_invalid, but this is neither intuitive nor documented. If only retry_on_empty is enabled and there is no response from the modbus unit, then transaction.py just logs "Retry on empty - 4" but doesn't actually perform a retry. This was broken by 67ab76adfa612c26cbadb2606f31caaebe5ead58 by @wildstray, reverting this commit solves the issue. I also see that I can reproduce this as far back as v2.4.0 de3955e46087f5cdcd77184da9a681d7ad20c032 , but does not reproduce in 2.3.0 fbdc470ae3e138c50e3659ec4ec8ebf39df58936

This issue is also mentioned in #488, but the original reporter and the follow up comments talk about two different things. The original report just sets things wrong, while the comments do talk about this, but don't get to the root of the issue.

### Code and Logs

In 2.3.0 with retry_on_empty=True:
```
2020-10-26 16:03:43,049 - INFO - util.ms_input - Reading holding from address 80, with length 16 from unit with ID 1
2020-10-26 16:03:43,049 - DEBUG - pymodbus.transaction - Current transaction state - TRANSACTION_COMPLETE
2020-10-26 16:03:43,049 - DEBUG - pymodbus.transaction - Running transaction 20
2020-10-26 16:03:43,049 - DEBUG - pymodbus.transaction - SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x50 0x0 0x10
2020-10-26 16:03:43,050 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-26 16:03:43,800 - DEBUG - pymodbus.transaction - Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2020-10-26 16:03:43,800 - DEBUG - pymodbus.transaction - RESETTING Transaction state to 'IDLE' for retry
2020-10-26 16:03:43,800 - DEBUG - pymodbus.transaction - Retry on empty - 3
2020-10-26 16:03:44,153 - DEBUG - pymodbus.transaction - SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x50 0x0 0x10
2020-10-26 16:03:44,154 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-26 16:03:44,905 - DEBUG - pymodbus.transaction - Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2020-10-26 16:03:44,905 - DEBUG - pymodbus.transaction - RESETTING Transaction state to 'IDLE' for retry
2020-10-26 16:03:44,905 - DEBUG - pymodbus.transaction - Retry on empty - 2
2020-10-26 16:03:45,532 - DEBUG - pymodbus.transaction - SEND: 0x0 0x14 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x50 0x0 0x10
2020-10-26 16:03:45,532 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-26 16:03:46,222 - DEBUG - pymodbus.transaction - Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-10-26 16:03:46,222 - DEBUG - pymodbus.transaction - RECV: 0x0 0x14 0x0 0x0 0x0 0x23 0x1 0x3 0x20 0x0 0x0 0x0 0x2a 0x0 0x0 0x0 0x2b 0x0 0x0 0x0 0x2c 0x0 0x0 0x0 0x2d 0x0 0x0 0x0 0x2e 0x0 0x0 0x0 0x2f 0x0 0x0 0x0 0x30 0x0 0x0 0x0 0x31
2020-10-26 16:03:46,223 - DEBUG - pymodbus.transaction - Adding transaction 20
2020-10-26 16:03:46,223 - DEBUG - pymodbus.transaction - Getting transaction 20
2020-10-26 16:03:46,223 - DEBUG - pymodbus.transaction - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-10-26 16:03:46,223 - INFO - util.ms_input - Read register with values [0, 42, 0, 43, 0, 44, 0, 45, 0, 46, 0, 47, 0, 48, 0, 49]
```

In 2.4.0 with retry_on_empty=True:
```
2020-10-26 18:32:36,454 - INFO - util.ms_input - Reading holding from address 80, with length 16 from unit with ID 1
2020-10-26 18:32:36,454 - DEBUG - pymodbus.transaction - Current transaction state - TRANSACTION_COMPLETE
2020-10-26 18:32:36,454 - DEBUG - pymodbus.transaction - Running transaction 11
2020-10-26 18:32:36,454 - DEBUG - pymodbus.transaction - SEND: 0x0 0xb 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x50 0x0 0x10
2020-10-26 18:32:36,454 - DEBUG - pymodbus.transaction - Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-10-26 18:32:37,205 - DEBUG - pymodbus.transaction - Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)) 
2020-10-26 18:32:37,206 - DEBUG - pymodbus.transaction - Retry on empty - 4
2020-10-26 18:32:37,206 - DEBUG - pymodbus.transaction - Getting transaction 11
2020-10-26 18:32:37,206 - DEBUG - pymodbus.transaction - Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-10-26 18:32:37,206 - ERROR - util.ms_input - IO error reading register with start decimal 80 and length 16 from unit with ID 1, is_socket_open: False, msg: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
```

dhoomakethu, 2020-12-19:
@thormick Could you please give a try with https://github.com/riptideio/pymodbus/tree/2.5.0

dhoomakethu, 2020-12-30:
Fixed with 2.5.0.
Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
Refer https://github.com/riptideio/pymodbus/releases

thormick, 2021-01-05:
Can confirm that this fixes the issue.

Sorry about becoming one of the `_no_response_devices`, my plans to hack on this project between Christmas and New Year fell through.

closed 2021-03-03 by dhoomakethu