pymodbus-dev/pymodbus #553: How to read the message which is being sent to modbus device in order to receive data from it ?

Abbad-Zafar, 2020-10-29:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

Abbad-Zafar, 2020-10-29:
@dhoomakethu

dhoomakethu, 2020-10-29:
Can you be more specific please?, I am not quite getting the question.

Abbad-Zafar, 2020-10-29:
I want the hex response or packet which the master sends to slave or vice versa

l337quez, 2020-10-30:
I would like to know how to read the complete message including reading the unit id

dhoomakethu, 2020-10-30:
@Abbad-Zafar there is no direct way at present to get raw response. You may have to reverse engineer the registers to get the frame.

Abbad-Zafar, 2020-11-03:
@dhoomakethu Doesn't the logging show info about packets ? What are being sent and what are being received ?

dhoomakethu, 2020-11-03:
You can view the data if you running with level debug.

nhi-vanye, 2020-11-25:
A spin out RFE for Serial would be to use `serial.serial_for_url()` instead of `serial.Serial()` to allow the `spy://` url schema to be used.

https://pyserial.readthedocs.io/en/latest/url_handlers.html

closed 2021-03-08 by github-actions[bot]