pymodbus-dev/pymodbus #559: Explore aioserial as a replacement for pyserial-asyncio

dhoomakethu, 2020-11-06:
### Versions

* Python: 3.6 and above
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```
https://github.com/changyuheng/aioserial

swamper123, 2020-11-06:
I like the way the code is written in modern terms. It looks uniform in each point. They have a point about high-level usage, which may help us a lot.

The code itself looks like "sync" async way to integrate pyserial into an async context. Only one read/write request is allowed and they are executed in an seperate Thread. I am not sure if this may lead in complications, since this way is not Threadsafe (trying read and write at the same time or (perhaps? not sure) receiving wrong messages).
I don't know at the moment how raising errors are handled in such cases, if the behaviour is normal or the Coroutine cancelled or what happens.

Also the code contains no tests (not github actions, no travis or similar). In my opinion this condition shall be minimum, before we integrated this in pymodbus 3.x

dhoomakethu, 2020-11-06:
Thanks for the inputs @swamper123 .  I am working on a bunch of transaction related enhancements which covers invalid responses, delayed responses agnostic to sync and async versions . This is one of the package I was hoping to explore as part of the exercise. Will keep posted on the findings.

closed 2021-03-08 by github-actions[bot]