pymodbus-dev/pymodbus #560: Pymodbus 3.x

dhoomakethu, 2020-11-06:
### Versions

* Python: 3.6 and above (3.7 and above preferrably)
* OS: linux|windows|osx
* Pymodbus: 3.x
* Modbus Hardware (if used):  NA

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

This issue covers a broader range of enhancements and bug fixes that will be going part of pymodbus 3.x which targets only python 3.6 and above. The support to python2.x will no longer be provided after the first release of 3.x.

### New features

1. Asyncio Server REPL with the ability to alter response dynamically (error response, delayed, chunked response etc)
2. Asyncio serial server

### Enhancements and bug fixes

1. Better handling of chunked response/incomplete/no response in synchronous clients
2. Validating incoming response against request , at present only the unit id and response length is validated and not the response function code
3. Better retry strategy on empty response. (Check recv buffer before sending the request again ??)
4. Ability to raise exceptions instead of returning `ModbusIOException` on error .
5. Asynchronous client better error handling on disconnections and re-connections
6. Functional tests 
7. Better suppport on windows (Volunteers required!!)
8. Increased performance and speed (???)
9. .....

Please feel free to comment with other enhancements and bugfixes which you think would be a good addition.

JamesJeffryes, 2021-02-17:
Thanks for all your work on this project! Python 3.10 is [scheduled to be released ](https://www.python.org/dev/peps/pep-0619/) in October of this year and will be the first version that will be incompatible with pymodbus 2.X code (due to the removal of the `@coroutine` decorator. I have several libraries that depend on this project that I would like to be able to update just as soon as we have 3.X code available so what can I do to help?

dhoomakethu, 2021-02-17:
@JamesJeffryes I am particularly held up with cleaning up . I have no definite timeline in mind to release 3.x soon. There is a branch [3.0.0](https://github.com/riptideio/pymodbus/tree/3.0.0) available which still python2 compatible . You can also look in to existing PR's and issues with label `3.x` which are required to move in to that direction.

swamper123, 2021-12-08:
Since this issue is now ~ a year old: Is it still goal to support 3.6 or shall we step up to 3.7 as minimal version?
IMO python 3.6 was a big step, especially for async stuff, but is still not so compatible with newer versions.

couling, 2022-04-13:
Python 3.6 is now EOL. https://endoflife.date/python though it wasn't technically at the time of @swamper123 's comment.
May I humbly suggest pymodbus 3 should drop support for all EOL versions.

swamper123, 2022-04-13:
@couling We are already a step ahead. with #736 we dropped the tests. ;)
But in general I am with you about EOL versions.

roguh, 2022-04-29:
`@coroutine` will actually be removed in Python 3.11, not 3.10

https://docs.python.org/3/library/asyncio-task.html

> Deprecated since version 3.8, will be removed in version 3.11: Use [async def](https://docs.python.org/3/reference/compound_stmts.html#async-def) instead.

janiversen, 2022-04-29:
and btw. @corutine works in 3.10 (at least here on my mac).

swamper123, 2022-05-02:
@roguh Is there still a leftover I have missed?
@janiversen Like rough said, it will be removed in 3.11. Since 3.8 it is deprecated.

janiversen, 2022-05-02:
Did any of you bother to look at the code? the @asyncio.coroutine is in line 24 of asyncio_test_helper.py, in a comment, so I really wonder what the talk is about.

janiversen, 2022-05-26:
Can we close this issue ?

dhoomakethu, 2022-06-04:
Yes, closing

closed 2022-06-04 by dhoomakethu