pymodbus-dev/pymodbus #565: concurrent.futures._base.TimeoutError with AsyncModbusTCPClient

ddabush, 2020-12-02:
### Versions

* Python:3.7
* OS:mac
* Pymodbus:  2.4
* Modbus Hardware (if used):  PLC

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

I am using AsyncModbusTCPClient in order to work with my hardware in asynchronous manner.
when I am using it and try to read data from registers I get TimeoutError.

but when I am using the blocking modbus client (from pyModbusTCP.client import ModbusClient) I am  able to read the data. what can cause it?

And also when debugging I see that under protocol it is written : Null Transport

### Code and Logs
from pymodbus.client.asynchronous.tcp import AsyncModbusTCPClient as ModbusClient
from pymodbus.client.asynchronous import schedulers
import asyncio

async def start_async_test(client):
    r = await client.read_holding_registers(0, 100)
    print(r)

loop = asyncio.new_event_loop()
asyncio.set_event_loop(loop)
new_loop, client = ModbusClient(schedulers.ASYNC_IO, port=502, loop=loop, host=ip, timeout=20)
loop.run_until_complete(start_async_test(client.protocol))
loop.close()

Error:

  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

 # code and logs here.

```

swamper123, 2020-12-03:
Did you meassure/verify that 20ms is enough time to respond?
Make a Wireshark trace (or similar) to verify that the packages arrive on time.

ddabush, 2020-12-03:
the timeout is on seconds:
```
  :param timeout: Time out in seconds
```
moreover the timeout is not related to it event if I change the timeout to 20000 I get timeout very fast.
```
Traceback (most recent call last):
  File "/Users/david.dabush/dev/csr_backend/services/motion_base/lib/motion_logic/sequencing/asyncio_poc.py", line 24, in start_async_test
    r = await client.read_holding_registers(0, 100)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

ddabush, 2020-12-03:
> Did you meassure/verify that 20ms is enough time to respond?
> Make a Wireshark trace (or similar) to verify that the packages arrive on time.
> 
> If the package needs more time, you either have to accept a longer timeout time or the blocking methods.
> If not, we may have to take a deeper look in the code (but I don't think so).

swamper123, 2020-12-03:
? What was the solution?

ddabush, 2020-12-03:
> ? Was that the solution?

I closed by mistake. the issue exists.

dhoomakethu, 2020-12-04:
@ddabush have you given a try with the latest dev branch ? It looks like this might have been fixed in there.

ddabush, 2020-12-04:
> @ddabush have you given a try with the latest dev branch ? It looks like this might have been fixed in there.

This is not an official release. I do I use it in my project?

swamper123, 2020-12-08:
> This is not an official release. I do I use it in my project?

If you got a requirements.txt in your project you can add:

`-e git+https://github.com/riptideio/pymodbus.git@dev#pymodbus`

Or installing it in your project like:
`python -m pip install git+https://github.com/riptideio/pymodbus.git@dev#pymodbus`

Both ways should (hopefully) work.

ddabush, 2020-12-08:
Still having this issue attaching picture of the connection while debugging:
[https://files.slack.com/files-pri/T0J3YV3RT-F01GEGTFL7M/screen_shot_2020-12-08_at_16.13.02.png](url)

dhoomakethu, 2020-12-08:
@ddabush OK will take a look and update

swamper123, 2020-12-09:
@ddabush You can copy paste the picture into github as well (I do that with snapshots). So everybody can have a look on it, instead of slack users only. ;)

ddabush, 2020-12-09:
> @ddabush You can copy paste the picture into github as well (I do that with snapshots). So everybody can have a look on it, instead of slack users only. ;)

Sorry :)
[image]
And for some reason the field of the host is 127.0.0.1 even though I gave it a value in the constructor.

swamper123, 2020-12-09:
In your code you provided code, you just have:

`host=ip`
as parameter, but ip is not defined, but I guess this is just a placeholder in that case ;)

But I may have found a clue!

```
loop = asyncio.new_event_loop() # <-- get a loop
asyncio.set_event_loop(loop) # <--- set the loop now
new_loop, client = ModbusClient(schedulers.ASYNC_IO, port=502, loop=loop, host=ip, timeout=20) # <-- returns a NEW LOOP! (where the client may be "active"?)
loop.run_until_complete(start_async_test(client.protocol)) # <--- not used NEW LOOP which was provided from above
loop.close()

```

Replace the `loop.run_until` .... with `new_loop.run_until .`... give it a shot, otherwise a shot for me 🍺

ddabush, 2020-12-09:
> In your code you provided code, you just have:
>
> `host=ip`
> as parameter, but ip is not defined, but I guess this is just a placeholder in that case ;)
>
> But I may have found a clue!
>
> ```
> loop = [...]

still having the timeout issue.
For some reason the ip and the timeout are not changing to the values I put in the constructor:

code:
```
from pymodbus.client.asynchronous.tcp import AsyncModbusTCPClient as ModbusClient
from pymodbus.client.asynchronous import schedulers
import asyncio

async def start_async_test(client):
    r = await client.read_holding_registers(0, 100)

# modbus_ip_tp1 = "10.11.31.98"
modbus_ip_tp2 = "10.11.31.97"
modbus_ip_tr2 = "10.11.31.89"
modbus_ip_tp3 = "10.11.31.99"

loop = asyncio.new_event_loop()
new_loop, client = ModbusClient(schedulers.ASYNC_IO, port=502, loop=loop, host=modbus_ip_tr2, timeout=20000)
connection = client.protocol
# connection.timeout = 20000
# connection.host = modbus_ip_tr2
asyncio.set_event_loop(new_loop)
new_loop.run_until_complete(start_async_test(connection))
loop.close()
```

the connection instance during debug:
[image]

Error:
```
  File "/Users/david/dev/venv/lib/python3.7/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

daviddabush, 2020-12-15:
@swamper123 I think if we will do short zoom and I will show you the problem we can make a real progress.

dhoomakethu, 2020-12-19:
@ddabush You are not passing `unit_id` in the the request . Can you pass `unit_id`. That should resolve this issue.

closed 2020-12-19 by dhoomakethu

daviddabush, 2021-03-03:
Hi,
Do you have modbus client for gevent?

dhoomakethu, 2021-03-03:
@daviddabush nope, but I believe you can extend the existing client implementations to support gevents.