pymodbus-dev/pymodbus #569: Forwarder does not respect the unit id when polling serial device

lampra1, 2020-12-03:
### Versions

* Python: 3.7.9
* OS: openwrt
* Pymodbus: 2.4
* Modbus Hardware (if used): Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

### Pymodbus Specific
* Forwarder example with 2 slave devices 
```python
    store = {0x1:RemoteSlaveContext(client),0x2:RemoteSlaveContext(client)}
    context = ModbusServerContext(slaves=store, single=False)         

```
### Description
I expect that when polling modbus tcp with device id 2 that the response will originate from the serial device with id 2 not from the serial device with id 1. This is not always the case.  The forwarder seems to respect the device id (i.e. does not handle requests for ids not in the store) but the serial poll does not.
I am not fluent with modbus but to me it seems that there is an issue when the modbus tcp request is translated to a serial request.
<details>
<summary>
### Correct response when polling device 1 & 2
</summary>
<p>

```python
## Device 1
##==========
DEBUG:pymodbus.server.sync:Started thread to serve client at ('10.200.200.8', 27146)
DEBUG:pymodbus.server.sync:Client Connected [10.200.200.8:27146]
DEBUG:pymodbus.server.sync:Handling data: 0x18 0xd4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1e 0x0 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x18 0xd4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1e 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
DEBUG:pymodbus.datastore.remote:validate[3] 30:1
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:SEND: 0x0 0x3 0x0 0x1e 0x0 0x1 0xe5 0xdd
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1607011134.845593, Current Time stamp - 1607011247.894504
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x2 0x0 0x1 0x79 0x84
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 0
DEBUG:pymodbus.transaction:Getting transaction 0
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
DEBUG:pymodbus.datastore.remote:get values[3] 30:1
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 6
DEBUG:pymodbus.transaction:SEND: 0x0 0x3 0x0 0x1e 0x0 0x1 0xe5 0xdd
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1607011247.932957, Current Time stamp - 1607011247.948632
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x2 0x0 0x1 0x79 0x84
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 0
DEBUG:pymodbus.transaction:Getting transaction 0
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
DEBUG:pymodbus.server.sync:send: [ReadRegisterResponse (1)]- b'18d4000000050103020001'

##Device 2
##=================
DEBUG:pymodbus.server.sync:Handling data: 0x18 0xd5 0x0 0x0 0x0 0x6 0x2 0x3 0x0 0x1e 0x0 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x18 0xd5 0x0 0x0 0x0 0x6 0x2 0x3 0x0 0x1e 0x0 0x1
DEBUG:pymodbus.factory:Factory Request[ReadHoldingRegistersRequest: 3]
DEBUG:pymodbus.datastore.remote:validate[3] 30:1
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
[... 138 lines cut ...]
##Usbmon dump
##===================
##Polling Device 1
ffffffc03dd8b800 2519222446 S Bo:002:02 -115 8 = 0003001e 0001e5dd
ffffffc03dd8b800 2519223467 C Bo:002:02 0 8 >
ffffffc03dd8ba00 2519253494 C Bi:002:02 0 7 = 01030200 017984
ffffffc03dd8ba00 2519253576 S Bi:002:02 -115 32 <
ffffffc03dd8b800 2519280816 S Bo:002:02 -115 8 = 0003001e 0001e5dd
ffffffc03dd8b800 2519281461 C Bo:002:02 0 8 >
ffffffc03dd8b200 2519313104 C Bi:002:02 0 7 = 01030200 017984
ffffffc03dd8b200 2519313144 S Bi:002:02 -115 32 <

##Polling Device 2
ffffffc03dd8b800 2527542186 S Bo:002:02 -115 8 = 0003001e 0001e5dd
ffffffc03dd8b800 2527542601 C Bo:002:02 0 8 >
ffffffc03dd8ba00 2527576601 C Bi:002:02 0 7 = 01030200 017984
ffffffc03dd8ba00 2527576681 S Bi:002:02 -115 32 <
ffffffc03dd8b800 2527597878 S Bo:002:02 -115 8 = 0003001e 0001e5dd
ffffffc03dd8b800 2527598569 C Bo:002:02 0 8 >
ffffffc03dd8b200 2527626602 C Bi:002:02 0 7 = 01030200 017984
ffffffc03dd8b200 2527626676 S Bi:002:02 -115 32 <

```
</p>
</details>

heidecjj, 2021-03-09:
For now, adding the `unit` kwarg to `RemoteSlaveContext` should give you the behavior you're expecting:

```python
    store = {0x1:RemoteSlaveContext(client, unit=1),0x2:RemoteSlaveContext(client, unit=2)}
    context = ModbusServerContext(slaves=store, single=False)   
```

lampra1, 2021-03-09:
@heidecjj Thank you, seems to be working.

PR #992 (2022-07-30): Add forwarder example with multiple slaves

fixes #569
closed 2022-07-30 by janiversen