pymodbus-dev/pymodbus #570: Server: ReportServerId can only customize the "ServerID" portion

karlp, 2020-12-03:
Function code 17 / 0x11 is poorly defined, (yay, modbus!)

[image] 
Some real world devices use a single byte for the server id, and XXX bytes for the "Additional data" and others use XXX bytes for the server id and have no additional data.

Currently, via undocumented properties set on a server context, users can customize the "Server ID" portion.  See https://github.com/riptideio/pymodbus/pull/562 and https://github.com/riptideio/pymodbus/pull/568

Ideally, users could customize the "Additional data" portion as well.  It would be straightforward to simply add another internal property on the context, but it seems like it might be time to think about it more thoughtfully.

Should the data used for ReportServerId be added to IModbusSlaveContext, as a new key on the datastore dict?  This becomes quite a pervsavive change, but it does make all the "slave data" be managed in the same place, and would make it easier to document as well.  At present, without setting properties on the context, you get a single information block from the _Server_ (not the unit/slave!) and so it's shared across all units.  [1]

Doing so would also make it a little more "expected" when it comes to spare data stores for instance.  At present, you can't disable function code 17 support, but if you provided an empty set explicitly, you could ensure that your modbus server ignored that request.

Because of the connected / overlapping issues here, I didn't want to try and work on a complete PR at this stage, so am opening this ticket for discussion / record keeping per discussions on gitter

[1] This also applies to the MEI structured blocks 43 / 0x2b, they _should_ be configurable per unit context, not once for the entire server.

karlp, 2020-12-03:
Why do I care?  Basically, implementing replicas of real world devices in pymodbus, so I don't need the actual hardware wired up.  Part of that includes all these corner functions.

dhoomakethu, 2020-12-30:
Fixed with 2.5.0.
Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
Refer https://github.com/riptideio/pymodbus/releases

closed 2021-03-03 by dhoomakethu