pymodbus-dev/pymodbus #572: ReadDeviceInformationRequest with ModbusSerialClient fails with struct.error

JulesEmangeard, 2020-12-15:
### Versions

* Python: 3.7.2
* OS: Windows 10
* Pymodbus: 2.4.0
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description
Hello,

I was trying to get Modbus RTU equipments informations (like brand or reference) using function code 43 (sub function 14) but when using the ReadDeviceInformationRequest with ModbusSerialClient I sometimes get a good answer and get the following error the rest of the time:
File "...\Python\Python37\lib\site-packages\pymodbus\mei_message.py", line 118, in calculateRtuFrameSize
    _, object_length = struct.unpack('>BB', buffer[size:size+2])

To fix this I changed _time.sleep(0.01)_ into _time.sleep(0.03)_ in _wait_for_data from class ModbusSerialClient at line 671 in sync.py.

### Code and Logs
In pymodbus/client/sync.py:

```python

    def _wait_for_data(self):
        size = 0
        more_data = False
        if self.timeout is not None and self.timeout != 0:
            condition = partial(lambda start, timeout:
                                (time.time() - start) <= timeout,
                                timeout=self.timeout)
        else:
            condition = partial(lambda dummy1, dummy2: True, dummy2=None)
        start = time.time()
        while condition(start):
            avaialble = self._in_waiting()
            if (more_data and not avaialble) or (more_data and avaialble == size):
                break
            if avaialble and avaialble != size:
                more_data = True
                size = avaialble
            time.sleep(0.03)
            # time.sleep(0.01)
```
Thank you for yout time,
Jules

dhoomakethu, 2020-12-15:
On Windows can you please initialize the client with `strict=False` and see if that fixes

JulesEmangeard, 2020-12-15:
I already did it but it didn't fixed it

dhoomakethu, 2020-12-16:
Thanks for the update. Unfortunately I don't have access to windows machine to try this. I will keep this open and see if there is a better way to pass wait time between waiting for data.

dhoomakethu, 2020-12-30:
Fixed with 2.5.0.
Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
Refer https://github.com/riptideio/pymodbus/releases

JulesEmangeard, 2020-12-31:
I installed the 2.5.0 version but unfortunately it didn't fix the problem.
When `time.sleep(0.1)` in `def _wait_for_data(self):` I get a response of my equipment around 2 times each 10 trys and when I set `time.sleep(0.2)` or `time.sleep(0.3)` then it's perfectly working.

dhoomakethu, 2020-12-31:
This is not yet handled , so the behaviour would be the same in 2.5.0

closed 2021-03-08 by github-actions[bot]