pymodbus-dev/pymodbus #573: Repl client single write register issue (resp.count)

ot2789, 2020-12-16:
Hello! I have a problem with the pymodbus library when writing a single register value. 
This is the error that I get:

**client.write_register(int(address), int(value), unit=self.__unit)**
  File "/usr/local/lib/python3.6/dist-packages/pymodbus/repl/client.py", line 50, in _wrapper
    'count': resp.count

Problem is in this file: https://github.com/riptideio/pymodbus/blob/master/pymodbus/register_write_message.py
The singular write register classes do not define the .count field inside them.

dhoomakethu, 2020-12-19:
@ot2789 what version of pymodbus are you using ? Can you share the command the you are using which resulted in this error?

dhoomakethu, 2020-12-19:
@ot2789 please give a try with `2.5.0` and let me know if that fixes issue.

ot2789, 2020-12-23:
> @ot2789 what version of pymodbus are you using ? Can you share the command the you are using which resulted in this error?

I am using 2.4.0. So the last released version. I will update to 2.5.0 to see if it is fixed.

ot2789, 2020-12-23:
> @ot2789 please give a try with `2.5.0` and let me know if that fixes issue.

Is there a way to install this via PIP? Or do I need to do the manual installation?

dhoomakethu, 2020-12-30:
@ot2789 sorry for the delay, try with `pip install git+https://github.com/riptideio/pymodbus.git@2.5.0`

dhoomakethu, 2020-12-30:
Fixed with 2.5.0rc1. Please install from https://pypi.org/project/pymodbus/2.5.0rc2/ .
Refer https://github.com/riptideio/pymodbus/releases

ot2789, 2021-01-04:
Hello! Yes the newest version fixed the problem! Thanks alot!

closed 2021-01-04 by ot2789