pymodbus-dev/pymodbus #576: Reverse TCP scheme question

mainakae, 2020-12-23:
### Versions

* Python: 3.9
* OS: raspberry / macOS
* Pymodbus: 2.4
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp - sync/async
* Client: tcp - sync/async

### Description

Hi everyone, and thanks for this awesome piece of software. First of all, I wanted to make clear that this is not an issue but more a question or support request. I didn't know where else to ask (google groups wouldn't let me and the gitter chat didn't seem the most appropriate way to do so).

My question is as follows: I've found myself in a number of times, in the situation where the Modbus TCP communication is started by the slave device. This makes the TCP scheme "reverse", that is: the Master (or traditional client) is waiting for a connection from a Slave (or traditional server).

I haven't found no clue on how to do this, nor in pymodbus, nor in a number of nodejs libraries I've been working so far for other projects; to the point of having even started in several occasions, the work of implementing it myself for one of the most used nodejs modbus libraries.

I'll try to explain why this "reversed TCP" scheme is of interest: let's imagine a slave device, say a power meter, that is installed in a remote location. In order to be able to connect to the device, the server needs an IP, but those devices are, more often than not, behind NATed networks, specially in the case of using SIM cards and dongles to connect (really frequent) where CG-NAT is really restrictive. Thus, the device has to initiate the connection to the server, which is a well known address and a beefy server waiting for those kind of connections. Once started the TCP connection the remaining interactions are initiated by the Master (the cloud server) and the Slave dutifully answers). The last time I've found myself in this situation has come from my solar panels installation at home, where the inverter just behaves that way (albeit using ethernet and my home router, instead of a SIM card).

It seems so utterly estrange to me by now, that there is no documentation, nor similar cases on the internet, that I start to think that IT DOES exist, and it has a name, which I don't know, and I'm just making an **** of myself :)

Any help/suggestion/hint in this regard or direction will be MUCH appreciated.

Thanks in advance, and best regards.

Pedro.

dcneeme, 2020-12-24:
Hi, are you thinking of unencrypted ModbusTCP connections over
      the public internet? Sounds like an unsafe/vulnerable thing to
      avoid... My suggestion would be: use a small local collector host
      to query your meters and forward the data in a secure way. 
    
    Neeme

dhoomakethu, 2020-12-24:
Can you give an example of a slave which offers this feature ?

mainakae, 2020-12-24:
Thanks a lot for commenting @dcneeme, it sure does sound unsafe but some ISP providers offer private networks via SIM, and also are affected by ECG-NAT, thus the "unsafeness" is somewhat mitigated. Anyway, adding TLS on top of the TCP connection is possible. The main point on this post is the reverse-ness of the situation in which the slave initiates the connection.

Best regards!
Pedro

mainakae, 2020-12-24:
Sure @dhoomakethu !, solar inverters from Huawei, network analyzers from ARELSA for smart lighting in smart city solutions, IPeX devices from Odin Solutions (also for sensoring and network analyzing on smart city scenarios)...

dhoomakethu, 2020-12-25:
OK, so if you want to do something similar with pymodbus there is no out of the box solution available, however if you are using a gateway like raspberry pi,  You can collect the data from the slave and push it to your backend over http/websocket/firebase/mqtt or whatever the solution you deem is good. With this approach you can be sure that only you have the access to the device and also monitor the data. May be setup a reverse ssh tunnel for you gain the access to the edge device for quick hacks and etc. A good starting point would be https://github.com/riptideio/pymodbus/blob/master/examples/common/updating_server.py and also https://github.com/riptideio/pymodbus/blob/master/examples/contrib/serial_forwarder.py

mainakae, 2020-12-26:
Yes, that is definitely possible, and saner to do than directly communicating through an insecure tcp channel. The case for insecure communications, though, is not that much: for instance, the SUN2000 solar inverters I'm now interfacing with, do use TLS on the TCP connection they establish. They even provide means to establish and check certificates for both the device and the server, so that communication cannot be that easily tampered with.

On the other hand, using a raspbery pi as a middle-man between the device you need to control, and the central management system, has another host of problems; such as the raspberry pi reliability (another piece that can  fail) and the fact that it will require extra development probably in both sides (the middle-man and the central MS end).

I do agree with you in the general feeling that ModBus should not be used in this way, though, but the thing is that it is being used as such, and we still need tools to cope with it.

closed 2021-03-08 by github-actions[bot]