pymodbus-dev/pymodbus #580: pymodbus\server\asynchronous.py\StartSerialServer: TypeError: __init__() got an unexpected keyword argument 'timeout'

shishir-dey, 2021-01-02:
### Versions

* Python: 3.7
* OS: Windows 10
* Pymodbus: 
* Modbus Hardware (if used): USB TTL

### Pymodbus Specific

* Server: rtu - async

### Description

The program crashes.

Seems like the ```twisted.internet.serialport.SerialPort``` does not have a ```timeout``` parameter. See https://twistedmatrix.com/documents/15.4.0/api/twisted.internet.serialport.SerialPort.html

### Code and Logs

Code:
```python
StartSerialServer(self.context, identity=self.identity,
                        port=GetSerialPort(), framer=ModbusRtuFramer, baudrate=38400)
```

Logs:
```python
2021-01-02 12:14:25,274 Thread-1        INFO     asynchronous   :327      Starting Modbus Serial Server on COM8
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\x\AppData\Local\Programs\Python\Python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\x\Downloads\Modbus Slave\tick.py", line 15, in run
    self.callback()
  File "C:\Users\x\Downloads\Modbus Slave\app.py", line 183, in modbus_server_task
    port=GetSerialPort(), framer=ModbusRtuFramer, baudrate=38400)
  File "C:\Users\x\AppData\Local\Programs\Python\Python37\lib\site-packages\pymodbus\server\asynchronous.py", line 340, in StartSerialServer
    rtscts=rtscts, bytesize=bytesize)
TypeError: __init__() got an unexpected keyword argument 'timeout'
```

dhoomakethu, 2021-01-02:
This should be fixed with 2.4.0. what version of pymodbus is this

dhoomakethu, 2021-01-02:
You can also give a try with 2.5.0rc2

jhfatehi, 2021-01-14:
I am running pymodbus 2.4.0 and got the same error.  I resolved it by removing the 'timeout' kwarg from line 339 in pymodbus/server/asynchronous.py.

shishir-dey, 2021-01-14:
> I am running pymodbus 2.4.0 and got the same error. I resolved it by removing the 'timeout' kwarg from line 339 in pymodbus/server/asynchronous.py.

I tried a similar workaround and the server works fine after that. Pymodbus version is 2.4.0.

dhoomakethu, 2021-01-14:
Did you guys tried 2.5.0rc2?

cti-jsweet, 2021-02-19:
@dhoomakethu checked this issue that I was having and it looks like the 2.5.0rc2 fixed the issue for me

cti-jsweet, 2021-02-19:
Doing some testing today on 2.5.0rc2 I'll try to update with my results.

dhoomakethu, 2021-03-02:
@shishir-dey I am not able to reproduce this issue, could you please paste the output of `pip freeze` of your python environment?

closed 2021-03-03 by dhoomakethu