pymodbus-dev/pymodbus #581: Add Trio support

altendky, 2021-01-02:
I am interested in having PyModbus support Trio async in addition to asyncio, Twisted, and Tornado.  No promises I will succeed but I have started exploring the code and am trying to piece together some amount of Trio functionality.  I figured it would be good to publicly document the interest and whatever efforts I make in case anyone else is interested or knows anything on the topic.

In case there is curiosity about my context, my particular use will be for [SunSpec](https://sunspec.org/) communication within a Qt Trio application using [QTrio](https://github.com/altendky/qtrio).  There are [pySunSpec](https://github.com/sunspec/pysunspec) and [pySunSpec2](https://github.com/sunspec/pysunspec2) but given that I wasn't able to get any interest from them in my working Twisted support several years ago I hesitate to invest again.  I'm not sure why they wrote their own Modbus anyways...  My expectation is to develop PyModbus Trio support and then add a SunSpec layer on top of that using pySunSpec2's data model but not its communication code.

altendky, 2021-03-03:
The PR is in work and waiting on others to get merged so as to have responsive CI.  Also, I've been on other stuff for a bit.

altendky, 2021-04-04:
Still running around on other tasks and haven't made it back here yet.

altendky, 2021-05-05:
Well, the PR has finally been submitted for review so...  `:]`  (open sesame)

altendky, 2021-06-10:
meh...

dhoomakethu, 2021-06-16:
Adding label so that this wont get stale.