pymodbus-dev/pymodbus #601: How can I read the registers

SayedMosalem, 2021-02-03:
https://pymodbus.readthedocs.io/en/latest/source/example/async_tornado_client_serial.html

How can I read the registers in the above example?

dhoomakethu, 2021-02-04:
Under `beginAsynchronousTest ` refer line ` rr = client.read_holding_registers(1, 1, unit=UNIT)`.

SayedMosalem, 2021-02-04:
> Under `beginAsynchronousTest ` refer line ` rr = client.read_holding_registers(1, 1, unit=UNIT)`.

I understand, I mean how can I access the value of the register @ the specified address. What "read_holding_register" is returing and how can I get specific info from the frame/response returned

Thanks for the quick reply

dhoomakethu, 2021-02-04:
ah sorry, you can access the registers with `rr.registers` . Please make sure you check for errors before hand with `rr.isError()`.

```
rr = client.read_holding_registers(0, 3, unit=1)
if not rr.isError():
    print(rr.registers)
else:
   print(f"Error - {rr}")
```

SayedMosalem, 2021-02-04:
> ah sorry, you can access the registers with `rr.registers` . Please make sure you check for errors before hand with `rr.isError()`.
> 
> ```
> rr = client.read_holding_registers(0, 3, unit=1)
> if not rr.isError():
>     print(rr.registers)
> else:
>    print(f"Error - {rr}")
> ```

Sorry to bother you but I have tried many times to access (rr.isError) or (rr.registers) and I got below error:
'Future' object has no attribute 'isError'   --> in the first trial
'Future' object has no attribute 'registers'  --> in the second trial when I removed the error check "if"

dhoomakethu, 2021-02-05:
You can see `dassert` function in the top, which acts as a callback to the future. `future.result` and `future.exception` should  give the actual values/exceptions. 
What version of tornado are you using? And when you run the examples are you seeing any errors?
Please also help with the version of pymodbus and python version you are using.

closed 2021-03-13 by github-actions[bot]