pymodbus-dev/pymodbus #602: 502 port refused.

NagarjunaReddy1242, 2021-02-11:
root@Omega-6834:~# python modbus_ip.py
ERROR:pymodbus.client.sync:Connection to (192.168.0.200, 502) failed: timed out
ERROR:pymodbus.client.sync:Connection to (192.168.0.200, 502) failed: timed out
Traceback (most recent call last):
  File "modbus_ip.py", line 13, in <module>
    rr = client.read_holding_registers(0x0001,2,unit=2)
  File "/usr/lib/python2.7/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/lib/python2.7/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.200:502)]

program:
from pymodbus.client.sync import ModbusTcpClient
import time
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)
host = '192.168.0.200'
port = 502

client = ModbusTcpClient(host,502)
client.connect()
while True:
        rr = client.read_holding_registers(0x0001,2,unit=2)
        print(rr)
        #assert(rr.function_code < 0x80)     # test that we are not an error
        #print rr.registers
        time.sleep(5)
why i am getting 502 port refused error? but i am able to pinging server IPaddress. only port error comming.

swamper123, 2021-02-12:
> only port error comming.

Have you checked your firewall configuration? Perhaps the port is blocked.

dhoomakethu, 2021-02-12:
Make sure your slave is running on the IP and PORT you have mentioned. The error says the remote is not accepting the connection or the connection request is not going out.

closed 2021-02-12 by dhoomakethu

NagarjunaReddy1242, 2021-02-12:
how to read Modbus UDP data using pymodslave simulator.

**#from pymodbus.client.sync import ModbusTcpClient
import win_inet_pton
import time
from pymodbus.client.sync import ModbusUdpClient as ModbusClient

#from pymodbus.transaction import ModbusRtuFramer
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient('172.16.1.140')
l=client.connect()
print(l)
#client.write_coil(1, True)
result = client.read_holding_registers(0x0001,10,unit=0x01)
print('hello')
print(result)
    
client.close()**

i didn't get any output.
**output:**True
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0xa