pymodbus-dev/pymodbus #609: Memory leak in sync servers due to socketserver memory leak.

laundmo, 2021-03-02:
### Versions

* Python: 3.7, 3.8, 3.9

### Pymodbus Specific
Affects: 
- sync ModbusTCPServer
- sync ModbusUDPServer

### Description

When running any sync server based on the stdlib socketserver, a memory leak in socketserver can be observed. This is a known and resolved issue, which is not yet implemented in the current releases of python 3.7, 3.8 and 3.9.

Relevant python bug report: https://bugs.python.org/issue37193

The issue is that non-daemonic threads are permanently appended to a list that is never cleared. This causes a memory leak which consists of Thread objects.

This is easily fixable by setting daemon_threads to True for either the ThreadingTCP/UDP server or by moneky patching that value in the ThreadingMixIn those classes inherit from.

Monkey Patch:
```py
import socketserver
socketserver.ThreadingMixIn.daemon_threads = True
```

laundmo, 2021-03-02:
@dhoomakethu this is not a enhancement, its a bug report/fix. currently long-running instances of the threading based servers WILL crash due to out of memory issues.

dhoomakethu, 2021-03-02:
@laundmo This is not a bug in the pymodbus library per-say and the fix is anyhow deemed to be available with future patches of python. We can have the patch as a hot-fix/enhancement for the time being and later choose to remove it.

laundmo, 2021-03-02:
@dhoomakethu i would still argue this should be labeled as bug, since its definitely something that will currently, and for 3.7 always (as the last bugfix release was 2020-06-27), make serious usage of the sync servers basically impossible.

also, as i said, if you want to support 3.7 you cannot expect to be able to remove the fix later, as 3.7 wont recieve that bugfix.

commit e88c41d0 (2021-03-03): Fix #609 monkey patch to handle memory leaks in sync server
closed 2021-03-03 by dhoomakethu