pymodbus-dev/pymodbus #610: ModbusSerialClient read_coils failed

wcchh, 2021-03-03:
### Versions

* Python: 3.6
* OS: ubuntu 16
* Pymodbus: 2.4.0
* Modbus Hardware (if used): FATEK FBs-CB25

### Pymodbus Specific
* Server: PLC
* Client: rtu - sync

### Description

I wanna read (64) coils data from a PLC. But received "incomplete message received" error... 
Through another modbus helper, I found my output command is "00 01 00 00 00 40 3C 2B".
And the other modbus tools send "01 01 00 00 00 40 3D FA" and it works.
I don't know what 's the case.
How can I using pyModbus to send above command to read coils?

### Code and Logs

```python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient

client = ModbusClient(method='rtu', port='/dev/ttyS0', baudrate=19200, parity='N', bytesize= 8, stopbits= 1, timeout= 1)
result = None
rr = client.read_coils(0, 64, uint=1)
if not rr.isError():
    print('result', rr.bits)
else:
    print('failed')

```

dhoomakethu, 2021-03-03:
Can you paste the logs of your transactions?

wcchh, 2021-03-04:
Here's the output as I see...
```python
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x40 0x3c 0x2b
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 2 bytes (0 received)) 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 0
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
failed
```

dhoomakethu, 2021-03-04:
@wcchh can you give a try with `2.5.0` and see if that helps?

wcchh, 2021-03-08:
pymodbus           2.5.0rc2
Has the same status.

dhoomakethu, 2021-03-08:
@wcchh  I have tried to reproduce this with latest pymodbus (2.5.0) and couldn't reproduce. Please share the modbus spec sheet for the modbus slave.

Also, if possible can you try any other transaction (reading holding/input registers ) if available?
Here is the snippet from the example I have tried.
```
> client.read_coils address=0 count=64 unit=1
2021-03-08 10:01:22,610 MainThread      DEBUG    transaction    :138      Current transaction state - TRANSACTION_COMPLETE
2021-03-08 10:01:22,610 MainThread      DEBUG    transaction    :143      Running transaction 2
2021-03-08 10:01:22,610 MainThread      DEBUG    transaction    :268      SEND: 0x1 0x1 0x0 0x0 0x0 0x40 0x3d 0xfa
2021-03-08 10:01:22,610 MainThread      DEBUG    rtu_framer     :267      Changing state to IDLE - Last Frame End - None, Current Time stamp - 1615177882.610891
2021-03-08 10:01:22,615 MainThread      DEBUG    sync           :76       New Transaction state 'SENDING'
2021-03-08 10:01:22,615 MainThread      DEBUG    transaction    :282      Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2021-03-08 10:01:22,618 MainThread      DEBUG    transaction    :369      Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2021-03-08 10:01:22,618 MainThread      DEBUG    transaction    :292      RECV: 0x1 0x1 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x34 0x1d
2021-03-08 10:01:22,618 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x1 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2021-03-08 10:01:22,619 MainThread      DEBUG    factory        :266      Factory Response[ReadCoilsResponse: 1]
2021-03-08 10:01:22,619 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2021-03-08 10:01:22,619 MainThread      DEBUG    transaction    :448      Adding transaction 1
2021-03-08 10:01:22,619 MainThread      DEBUG    transaction    :459      Getting transaction 1
2021-03-08 10:01:22,620 MainThread      DEBUG    transaction    :221      Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
```

wcchh, 2021-03-08:
Thanks for your reply~

Would you please run my sample code (in python file) to check the output? I think my code cannot specify slave ID and only as reading coils.

If you cannot reproduce it, that is` SEND: 0x0 0x1`, then I think it might be some system issues...

My sample code： 
```python
def test_pyModbus():
    import logging
    logging.basicConfig()
    log = logging.getLogger()
    log.setLevel(logging.DEBUG)
    from pymodbus.client.sync import ModbusSerialClient as ModbusClient

    client = ModbusClient(method='rtu', port='/dev/ttyS0', baudrate=19200, parity='N', bytesize= 8, stopbits= 1, timeout= 5)
    result = None
    rr = client.read_coils(0, 64, UINT=1)
    if not rr.isError():
        print('result', rr.bits)
    else:
        print('failed')

test_pyModbus()
```
And the output of mine:
> DEBUG:pymodbus.transaction:Current transaction state - IDLE
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: **0x0** 0x1 0x0 0x0 0x0 0x40 0x3c 0x2b
> DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
> DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
> DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 2 bytes (0 received)) 
> DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
> DEBUG:pymodbus.transaction:Getting transaction 0
> DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
> failed

BTW, other transaction (reading holding registers ) seems good:
> DEBUG:pymodbus.transaction:Running transaction 1
> DEBUG:pymodbus.transaction:SEND: **0x1** 0x3 0x0 0x0 0x0 0x40 0x44 0x3a

dhoomakethu, 2021-03-08:
@wcchh You are passing `unit` wrong. It should be `rr = client.read_coils(0, 64, unit=1)` not `UINT=1`

wcchh, 2021-03-08:
Ah...~ so sorry.

closed 2021-03-08 by wcchh

janiversen, 2022-08-09:
Congratulations.