pymodbus-dev/pymodbus #613: Number of registers

Panther50500, 2021-03-07:
### Versions

* Python: 3.7
* OS: Raspberry OS
* Pymodbus: 2.5
* Modbus Hardware (if used): Raspberry Pi / FT232

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description

Hello, I wanted configure the size of the modbus strucure as i needed (32 words). May be it is a misunderstanding of me but when I create a modbus structure of 32 words on the server, I can read or write until only 31 words with the client (0 to 30).

Otherwise, I would like to thank everyone who contributes to this library, it is great and very functional.

Fred

### Code and Logs

My code :+1: 

#!/usr/bin/env python
from pymodbus.server.sync import StartSerialServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock, ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from pymodbus.transaction import ModbusRtuFramer, ModbusBinaryFramer
# --------------------------------------------------------------------------- #
# configure the service logging
# --------------------------------------------------------------------------- #
import serial

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s'
          ' %(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def run_server():
    **block = ModbusSequentialDataBlock(0, [0]*32)**
    slaves  = {
        0x01: ModbusSlaveContext(hr=block,ir=block)
    }
    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '2.3.0'

    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='/dev/ttyS0', stopbits = 1, bytesize = 8, parity = 'N', baudrate=19200, timeout=0.005)

if __name__ == "__main__":
    run_server()

On the client :+1: 

rr = self.client.read_holding_registers(0, 32, unit=0x01)

Log :+1: 

2021-03-07 19:56:07,597 MainThread      DEBUG    sync           :46       Client Connected [/dev/ttyS0:/dev/ttyS0]
2021-03-07 19:56:07,598 MainThread      DEBUG    sync           :580      Started thread to serve client
2021-03-07 19:56:10,688 MainThread      DEBUG    rtu_framer     :185      Getting Frame - 0x3 0x0 0x0 0x0 0x20
2021-03-07 19:56:10,688 MainThread      DEBUG    factory        :137      Factory Request[ReadHoldingRegistersRequest: 3]
2021-03-07 19:56:10,688 MainThread      DEBUG    rtu_framer     :107      Frame advanced, resetting header!!
2021-03-07 19:56:10,689 MainThread      DEBUG    context        :64       validate: fc-[3] address-1: count-32
2021-03-07 19:56:10,689 MainThread      ERROR    pdu            :107      Exception Response(131, 3, IllegalAddress)
2021-03-07 19:56:10,689 MainThread      DEBUG    sync           :144      send: [Exception Response(131, 3, IllegalAddress)]- b'018302c0f1'

Panther50500, 2021-03-07:
I wanted to put in bold this line : 

block = ModbusSequentialDataBlock(0, [0]*32) 

but it didn't display well

Fred

tracernz, 2021-03-07:
Try ModbusSparseDataBlock.

dhoomakethu, 2021-03-08:
Use `zero_mode=True` Refer the below comment from `synchronous_server` example.
```
# The slave context can also be initialized in zero_mode which means that a
    # request to address(0-7) will map to the address (0-7). The default is
    # False which is based on section 4.4 of the specification, so address(0-7)
    # will map to (1-8)::
    #
    #     store = ModbusSlaveContext(..., zero_mode=True)
```

closed 2021-03-08 by dhoomakethu

LemonCANDY42, 2021-05-17:
> Use `zero_mode=True` Refer the below comment from `synchronous_server` example.
>
> ```
> # The slave context can also be initialized in zero_mode which means that a
>     # request to address(0-7) will [...]

It's Work!