pymodbus-dev/pymodbus #614: [Errno 111] Connection refused

qistoph, 2021-03-09:
### Versions

* Python: 3.9.2
* OS: Alpine Linux v.3.13 (on Docker 19.03.13)
* Pymodbus: v.2.5.0
* Modbus Hardware (if used): Modbus TCP, no specific hardware

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?
Trying to connect to my SolarEdge SE3000H like on version v.2.4.0. Since v.2.5.0 however this always results in connection refused.

The weird thing however is that Wireshark/tcpdump shows a full SYN, SYN/ACK, ACK and even data transferred in both ways.

Reverting back (pymodbus==2.4.0) immediately solves the issue.

### Code and Logs

```
seinfluxfwd_1  | 2021-03-09 08:46:35,216 [pymodbus.client.sync] ERROR    Connection to (192.168.180.50
, 1502) failed: [Errno 111] Connection refused
seinfluxfwd_1  | 2021-03-09 08:46:35,216 [__main__    ] ERROR    Modbus Error: [Connection] Failed to 
connect[ModbusTcpClient(192.168.180.50:1502)]
```

```
09:47:49.870838 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [S], seq 1105518497, win 64240, options [mss 1460,sackOK,TS val 1785985442 ecr 0,nop,wscale 7], length 0
09:47:49.871287 IP 192.168.180.50.1502 > 192.168.178.7.35007: Flags [S.], seq 3895959532, ack 1105518498, win 28960, options [mss 1460,sackOK,TS val 146021264 ecr 1785985442,nop,wscale 4], length 0
09:47:49.871762 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [.], ack 1, win 502, options [nop,nop,TS val 1785985443 ecr 146021264], length 0
09:47:49.871845 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [P.], seq 1:13, ack 1, win 502, options [nop,nop,TS val 1785985443 ecr 146021264], length 12
09:47:49.872169 IP 192.168.180.50.1502 > 192.168.178.7.35007: Flags [.], ack 13, win 1810, options [nop,nop,TS val 146021264 ecr 1785985443], length 0
09:47:49.899171 IP 192.168.180.50.1502 > 192.168.178.7.35007: Flags [P.], seq 1:148, ack 13, win 1810, options [nop,nop,TS val 146021294 ecr 1785985443], length 147
09:47:49.899663 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [.], ack 148, win 501, options [nop,nop,TS val 1785985471 ecr 146021294], length 0
09:47:49.900306 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [F.], seq 13, ack 148, win 501, options [nop,nop,TS val 1785985471 ecr 146021294], length 0
09:47:49.900391 IP 192.168.178.7.40655 > 192.168.180.50.1502: Flags [S], seq 1860378655, win 64240, options [mss 1460,sackOK,TS val 1785985471 ecr 0,nop,wscale 7], length 0
09:47:49.900777 IP 192.168.180.50.1502 > 192.168.178.7.40655: Flags [R.], seq 0, ack 1860378656, win 0, length 0
09:47:49.906846 IP 192.168.180.50.1502 > 192.168.178.7.35007: Flags [F.], seq 148, ack 14, win 1810, options [nop,nop,TS val 146021304 ecr 1785985471], length 0
09:47:49.907309 IP 192.168.178.7.35007 > 192.168.180.50.1502: Flags [.], ack 149, win 501, options [nop,nop,TS val 1785985479 ecr 146021304], length 0
```

```python
        client = ModbusTcpClient(solaredge['ip'], solaredge['port'])
        ans = client.read_holding_registers(address=SUNSPEC_I_COMMON_BLOCK_START, count=69, unit=1)
        assert(not ans.isError())
```

dhoomakethu, 2021-03-09:
Can you paste the complete logs for pymodbus 2.5.0 and 2.4.0 please?

qistoph, 2021-03-09:
And the logs (forgot to enable debug for the module)

**2.4.0**
```
pymodbus_1  | 2021-03-09 21:24:17,931 [pymodbus.transaction] DEBUG    Current transaction state - IDLE
pymodbus_1  | 2021-03-09 21:24:17,931 [pymodbus.transaction] DEBUG    Running transaction 1
pymodbus_1  | 2021-03-09 21:24:17,931 [pymodbus.transaction] DEBUG    SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x40 0x0 0x45
pymodbus_1  | 2021-03-09 21:24:17,931 [pymodbus.client.sync] DEBUG    New Transaction state 'SENDING'
pymodbus_1  | 2021-03-09 21:24:17,931 [pymodbus.transaction] DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
pymodbus_1  | 2021-03-09 21:24:17,964 [pymodbus.transaction] DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
pymodbus_1  | 2021-03-09 21:24:17,964 [pymodbus.transaction] DEBUG    RECV: 0x0 0x1 0x0 0x0 0x0 0x8d 0x1 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x48 0x2d 0x52 0x57 0x30 0x30 0x30 0x42 0x4e 0x4e 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x30 0x38 0x2e 0x30 0x30 0x31 0x39 0x0 0x0 0x37 0x33 0x31 0x42 0x43 0x30 0x39 0x30
 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1
pymodbus_1  | 2021-03-09 21:24:17,964 [pymodbus.framer.socket_framer] DEBUG    Processing: 0x0 0x1 0x0 0x0 0x0 0x8d 0x1 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67
 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x48 0x2d 0x52 0x57 0x30 0x30 0x30 0x42 0x4e 0x4e 0x34 0x0 0x0 0x0 0x0 0x0 0
x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x30 0x38 0x2e 0x30 0x30 0x31 0x39 0x0 0x0 0x37 0x33 0x31 0x42 0x43
 0x30 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.factory] DEBUG    Factory Response[ReadHoldingRegistersResponse: 3]
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Adding transaction 1
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Getting transaction 1
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
pymodbus_1  | 2021-03-09 21:24:17,965 [__main__    ] DEBUG    b'SunS\x00\x01\x00ASolarEdge \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00SE3000H-RW000BNN4\x00\x00\
x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x000004.0008.0019\x00\x00731BC090\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01'
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Current transaction state - TRANSACTION_COMPLETE
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Running transaction 2
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x85 0x0 0x34
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.client.sync] DEBUG    New Transaction state 'SENDING'
pymodbus_1  | 2021-03-09 21:24:17,965 [pymodbus.transaction] DEBUG    Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
pymodbus_1  | 2021-03-09 21:24:17,983 [pymodbus.transaction] DEBUG    Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.transaction] DEBUG    RECV: 0x0 0x2 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x65 0x0 0x32 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xfe 0x8 0x9e 0xff 0xff 0xff 0xf
f 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0 0xc3 0x47 0xff 0xfd 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x31 0xa6 0x62 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xf 0xff 0xff 0x0 0x0 0x0 0x0 0x
80 0x0 0xa 0xc5 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x2 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.framer.socket_framer] DEBUG    Processing: 0x0 0x2 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x65 0x0 0x32 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xfe 0x8 0x9e 0xf
f 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0 0xc3 0x47 0xff 0xfd 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x31 0xa6 0x62 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xf 0xff 0xff 0x0
 0x0 0x0 0x0 0x80 0x0 0xa 0xc5 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x2 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x0 0x0 0x0 0x0
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.factory] DEBUG    Factory Response[ReadHoldingRegistersResponse: 3]
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.transaction] DEBUG    Adding transaction 2
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.transaction] DEBUG    Getting transaction 2
pymodbus_1  | 2021-03-09 21:24:17,984 [pymodbus.transaction] DEBUG    Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
[... cut ...]
pymodbus_1  |   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
pymodbus_1  |     return self.execute(request)
pymodbus_1  |   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 108, in execute
pymodbus_1  |     raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus_1  | pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(10.10.180.50:1502)]
pymodbus_pymodbus_1 exited with code 1
```

dhoomakethu, 2021-03-10:
@qistoph Thanks for the logs, on pymodbus 2.4.0, can you add `client.close()` after your reads and see if there is a change in the behaviour? Pymodbus 2.5.0 closes socket connection after every read explicitly and opens the connection on reads.

```
client = ModbusTcpClient(solaredge['ip'], solaredge['port'])
ans = client.read_holding_registers(address=SUNSPEC_I_COMMON_BLOCK_START, count=69, unit=1)
client.close()
assert(not ans.isError())

blob = pack(">69H", *ans.registers)
logger.debug(blob)

ans = client.read_holding_registers(address=SUNSPEC_I_MODEL_BLOCK_START, count=52, unit=1)
assert(not ans.isError())
blob = pack(">52H", *ans.registers)
logger.debug(blob)
```

qistoph, 2021-03-10:
That does replicate the issue in 2.4.0:

```
pymodbus_1  | 2021-03-10 05:53:24,031 [pymodbus.transaction] DEBUG    Current
 transaction state - IDLE
pymodbus_1  | 2021-03-10 05:53:24,031 [pymodbus.transaction] DEBUG    Running
 transaction 1
pymodbus_1  | 2021-03-10 05:53:24,031 [pymodbus.transaction] DEBUG    SEND: 0
x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x40 0x0 0x45
pymodbus_1  | 2021-03-10 05:53:24,031 [pymodbus.client.sync] DEBUG    New Tra
nsaction state 'SENDING'
pymodbus_1  | 2021-03-10 05:53:24,031 [pymodbus.transaction] DEBUG    Changin
g transaction state from 'SENDING' to 'WAITING FOR REPLY'
pymodbus_1  | 2021-03-10 05:53:24,059 [pymodbus.transaction] DEBUG    Changin
g transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
pymodbus_1  | 2021-03-10 05:53:24,059 [pymodbus.transaction] DEBUG    RECV: 0
x0 0x1 0x0 0x0 0x0 0x8d 0x1 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41 0x5
3 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x
0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30
 0x30 0x30 0x48 0x2d 0x52 0x57 0x30 0x30 0x30 0x42 0x4e 0x4e 0x34 0x0 0x0 0x0
 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x30 0
x38 0x2e 0x30 0x30 0x31 0x39 0x0 0x0 0x37 0x33 0x31 0x42 0x43 0x30 0x39 0x30
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0
x0 0x0 0x0 0x0 0x0 0x0 0x1
pymodbus_1  | 2021-03-10 05:53:24,059 [pymodbus.framer.socket_framer] DEBUG
  Processing: 0x0 0x1 0x0 0x0 0x0 0x8d 0x1 0x3 0x8a 0x53 0x75 0x6e 0x53 0x0 0
x1 0x0 0x41 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0
 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53
 0x45 0x33 0x30 0x30 0x30 0x48 0x2d 0x52 0x57 0x30 0x30 0x30 0x42 0x4e 0x4e 0
x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0
x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0
x30 0x30 0x30 0x38 0x2e 0x30 0x30 0x31 0x39 0x0 0x0 0x37 0x33 0x31 0x42 0x43
0x30 0x39 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x
0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1
pymodbus_1  | 2021-03-10 05:53:24,059 [pymodbus.factory] DEBUG    Factory Res
ponse[ReadHoldingRegistersResponse: 3]
pymodbus_1  | 2021-03-10 05:53:24,060 [pymodbus.transaction] DEBUG    Adding
transaction 1
pymodbus_1  | 2021-03-10 05:53:24,060 [pymodbus.transaction] DEBUG    Getting
 transaction 1
pymodbus_1  | 2021-03-10 05:53:24,060 [pymodbus.transaction] DEBUG    Changin
g transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
pymodbus_1  | 2021-03-10 05:53:24,060 [__main__    ] DEBUG    b'SunS\x00\x01\
x00ASolarEdge \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0\x00\x00\x00\x00\x00\x00SE3000H-RW000BNN4\x00\x00\x00\x00\x00\x00\x00\x00\x0
0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
\x00\x00\x000004.0008.0019\x00\x00731BC090\x00\x00\x00\x00\x00\x00\x00\x00\x0
0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01'
pymodbus_1  | 2021-03-10 05:53:24,061 [pymodbus.client.sync] ERROR    Connect
ion to (10.10.180.50, 1502) failed: [Errno 111] Connection refused
pymodbus_1  | Traceback (most recent call last):
pymodbus_1  |   File "/usr/src/app/test.py", line 40, in <module>
pymodbus_1  |     ans = client.read_holding_registers(address=SUNSPEC_I_MODEL
_BLOCK_START, count=52, unit=1)
pymodbus_1  |   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/
common.py", line 114, in read_holding_registers
pymodbus_1  |     return self.execute(request)
pymodbus_1  |   File "/usr/local/lib/python3.9/site-packages/pymodbus/client/
sync.py", line 107, in execute
pymodbus_1  |     raise ConnectionException("Failed to connect[%s]" % (self._
_str__()))
pymodbus_1  | pymodbus.exceptions.ConnectionException: Modbus Error: [Connect
ion] Failed to connect[ModbusTcpClient(10.10.180.50:1502)]
```

dhoomakethu, 2021-03-10:
@qistoph Thanks, so that basically indicates your slave has a tolerance to the number of immediate connects, probably a sleep between the reads should be a workaround on pymodbus 2.5.0. But I will see if I can add a support to disable disconnects after read as a seperate kwarg to client initialization.

qistoph, 2021-03-10:
Thanks! Adding a delay of 40ms indeed makes my code work with 2.5.0.

An option to disable disconnects sure would be nice.

dhoomakethu, 2021-03-11:
Please give a try with 2.5.1rc1 .
Release notes: https://github.com/riptideio/pymodbus/releases/tag/v2.5.1rc1
Pypi: https://pypi.org/project/pymodbus/2.5.1rc1/

qistoph, 2021-03-11:
2.5.1rc1 fixes my issue, without the sleep work-around :tada:

closed 2021-03-30 by dhoomakethu