pymodbus-dev/pymodbus #616: ModbusSerialClient socket closed after first read

thunderbug1, 2021-03-12:
I recently upgraded from python 2.7 to python 3.8.5
The code uses the Pymodbus library to connect to a ups and to periodically read some values.

In python2 this worked but with python3 the client says the port is closed after the first read (even though further reads work )

The connection is established like:

BAUDRATE=115200
STOP_BITS=1
START_BITS=1
BYTESIZE=8
PARITY="E"
SLAVE_ID = 192

self.client = ModbusSerialClient(method='rtu', port=self.ups_port, timeout=1, baudrate=BAUDRATE, stopbits=STOP_BITS, startbits=START_BITS, bytesize=BYTESIZE, parity=PARITY, )

self.client.connect() # this returns true

# Then I check the connection like:

connected = self.client.is_socket_open()
print(f"connect connected = {connected}")  # this prints "true"

rr = self.client.read_input_registers(0x4202,1, unit=SLAVE_ID)
if rr.isError():
            rospy.logerr("encountered pymodbus exception_code: {}".format(ModbusExceptions.decode(rr.exception_code)))  # no exceptions are encountered

connected = self.client.is_socket_open()
print(f"connect connected = {connected}")  # this prints "false"

dhoomakethu, 2021-03-12:
Please try `pymodbus==2.5.1rc1` or you can add a small sleep between your reads.

aetasoul, 2021-03-12:
https://github.com/riptideio/pymodbus/issues/612

https://stackoverflow.com/questions/66526532/after-read-holding-registers-with-pymodbus-connection-is-broken-in-raspberry-pi/66581700?noredirect=1#comment117707861_66581700

thunderbug1, 2021-03-17:
ok thank you very much for your answers, the fix with pymodbus==2.5.1rc1 resolved the issue for me :)

closed 2021-03-17 by thunderbug1