pymodbus-dev/pymodbus #617: Modbus Server generate many child processes

jalarconn, 2021-03-13:
Hi All, 

I have implemented a Modbus server with the PyModbus library.
For some time now, the process has been generating many child processes that I cannot control.
I would appreciate your help to optimize or control this process.
The server has modbus scalable units that write data and other units that query data.
Thanks

dhoomakethu, 2021-03-14:
@jalarconn can you please share more information ?Sample code, process tree etc?

jalarconn, 2021-03-14:
Yes, it's a code.

from pymodbus.server.sync import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock, ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer, ModbusBinaryFramer

def run_server():

    store = ModbusSlaveContext(0, zero_mode=True)    
    context = ModbusServerContext(slaves=store, single=True)
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = '2.3.0'
    StartTcpServer(context, identity=identity, address=('127.0.0.1', 502))

if __name__ == "__main__":
    run_server()

jalarconn, 2021-03-14:
[image]

This is the problem. The process automatically generates many child processes that I cannot control and end up using a lot of machine resources.
Thank to all

dhoomakethu, 2021-03-15:
Thanks, that's strange. What OS are you using?

jalarconn, 2021-03-15:
Hello, thanks for your answer, I am using Ubuntu Server 18.04

Atentamente,
Jaime Alacón N.

dhoomakethu, 2021-03-16:
Thanks, I donot have access to Ubuntu server so I couldn't test it there. But on Mac OSX I don't see such behaviour. What version of pymodbus and python is this?

closed 2021-05-06 by github-actions[bot]