pymodbus-dev/pymodbus #618: The same context for different slaves (SqlSlaveContext)

CzakuGim, 2021-03-15:
### Versions

* Python: 3.8.5
* OS: ubuntu 20.05
* Pymodbus: 2.5.0
* Modbus Hardware (if used): not yet

### Pymodbus Specific
* Server: rtu- async
* Client: rtu- sync

### Description

Hello,

I have some question about pymodbus and handling multicontext with sql.

I have found how to do single context handling using sql:

https://stackoverflow.com/questions/61949194/pymodbus-modbus-server-implementation-with-multiple-slave-devices-context-writ

So:

```
class CustomDataBlock(ModbusSparseDataBlock):
    def setValues(self, address, value):
        super(ModbusSparseDataBlock, self).setValues(address, value)
        print(f"wrote value: {value} to address: {address}")

def run_async_server(context):
    block = CustomDataBlock([0]*100)#ModbusSequentialDataBlock(0x00, [0] * 0xff)
    store  = SqlSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = version.short()

    # RTU Server
    StartSerialServer(context, identity=identity, 
                      port='/dev/pts/6', framer=ModbusRtuFramer, baudrate=115200)

if __name__ == "__main__":
    run_async_server()
```

And it's working. Of course if I will fill the sql database.

I have also tried to handle multiple context - one per slave_id. Take a look below:

```
    block = CustomDataBlock([0]*100)
    store  = SqlSlaveContext(di=block, co=block, hr=block, ir=block)
    slaves = {
        0x01: store,
        0x02: store
    }
    context = ModbusServerContext(slaves=slaves, single=False)
```

So, I have tried to get data using my testing slave (there I can write/read registers or coils).

If the slave id is 0x03 - works as expected - of course I dont have access to master context - it's handling only addresses 0x01 and 0x02.

But when I have used 0x02 or 0x01 - I have access to exactly the same data.

I was sure that the context will be different for slave with id 0x01 and for 0x02.

dhoomakethu, 2021-03-16:
Try `single=False` that should do the trick.

dhoomakethu, 2021-03-16:
Where is slave id `0x03` in your code? You are allocating the same block to both slaves, have you tried creating separate blocks and see if that helps?

closed 2021-04-22 by github-actions[bot]