pymodbus-dev/pymodbus #619: Unable to close sync ModbusTCPServer

TheBigFudge, 2021-03-16:
### Versions

* Python: 3.8
* OS: Ubuntu
* Pymodbus: 2.5.1rc1

### Pymodbus Specific
* Server: tcp - sync
* Client: tcp- sync

### Description

I'm currently building a Modbus server in a Docker image and extend the synchronous_server.py example for it:
[link](https://github.com/riptideio/pymodbus/blob/master/examples/common/synchronous_server.py)

Unfortunately I am not able to shutdown the server properly.I need this so that the Container can be terminated properly, for this I would like to terminate the server from another thread.

But unfortunately StartTcpServer doesn't return the server instance, so I can't execute a shutdown() or close_server() function.
How can I terminate serve_forever() and with it the Modbus server thread correctly?

Unfortunately I can't find a thread about it and the examples don't help either

dhoomakethu, 2021-03-17:
How are you shutting down the server at present? Are you usning `psutil` to grab the `pid` and terminate ? If `synchronous_server.py` is your primary process (with PID 1), you can just kill that but that would also terminate your container. Also make sure there are no clients connected to the server (just for testing).

TheBigFudge, 2021-03-17:
Hey @dhoomakethu 

Thanks for the quick reply. Currently I do not shutdown the server at all, so the container is not terminated yet. But that is my goal.

So basicly I have a main program (main.py) with PID 1 in the container, which opens a thread for the synchronous modbus server and starts the server. Now I want to terminate all processes when the container gets Ctrl+C, but unfortunately I can't find a way to send a stop signal from the main program (main.py) to the server thread. Unfortunately StartTcpServer does not return the server instance either.

Is there no function that terminates all client connections and terminates the server that I can trigger when the container gets Ctrl+C? `psutil `would maybe one way, but isn't there another "cleaner" way, so that the server can gracefully end itself?

TheBigFudge, 2021-03-17:
have replaced `StartTcpServer ` with `ModbusTcpServer` and now I can run .shutdown() on the server instance. I could have thought of that myself, sorry for the inconvenience

closed 2021-03-17 by TheBigFudge