pymodbus-dev/pymodbus #621: Given read request, unable to read multiple Modbus RTU responses from usb serial in Continuous/streaming mode

robotichustle, 2021-03-31:
### Versions

* Python: 3.6.9
* OS: Ubuntu 18.04.5 LTS
* Pymodbus: 2.5.0
* Modbus Hardware (if used): Forsentek Load cell indicator (http://www.forsentek.com/prodetail_288.html)

### Pymodbus Specific
* Server: rtu - sync
* Client: rtu - sync

### Description

Hi guys,

I am able to successfully read values from a Modbus device connected to a torque sensor using pymodbus.client.sync.ModbusSerialClient in RTU mode in Ubuntu Linux 18.04. The configuration: baudrate=57600, method='rtu', port='/dev/ttyUSB0', stopbits=1, bytesize=8, parity='E', timeout=10.

I can see that an average data acquisition frequency in the computer side is ~100 Hz. However, this is not enough for my application because I want to get the data at highest frequency as possible perhaps double i.e. ~200 Hz of instantaneous torque while the motor is moving.

I understand the standard Modbus RTU is a Request-Response communication protocol so in my main loop I have to keep sending a Request to get a Response so I can decode it.

Since the torque sensor also has a "Continuous mode" which auto-sends response packets at 10 ms intervals (configurable), I am thinking I could potentially increase the frequency of reading the values by only sending a Request ONCE (outside the main loop) and then keep reading the responses from the serial buffer for as long as I want to inside the main loop.

I don't know how to modify my code to achieve this using pymodbus. I suspect we need to modify the pymodbus.transaction.ModbusRtuFramer somehow. Or manually advanceFrame() somehow. Is this possible?

### Code and Logs

```python
from pymodbus.client.sync import ModbusSerialClient
import logging

def main():
    # Setup logger
    logging.basicConfig()
    log = logging.getLogger()
    log.setLevel(logging.DEBUG)

    # Initialise Modbus client
    client = ModbusSerialClient(
        method='rtu', port='/dev/ttyUSB0', stopbits=1, bytesize=8, parity='E', timeout=10, 
        baudrate=57600, # Maximum baudrate possible on the Modbus server device
    )
    if client.connect():
        # Successfully connected to Modbus Server/Slave device

        # NOTE: Send request only ONCE OUTSIDE main loop. Moving this INSIDE the main loop works flawlessly!
        res = client.read_holding_registers(address=0, count=2, unit=1)

        while True: # --- MAIN LOOP ----
            # HELP How do I keep decoding the next frame from the USB serial port without having to SEND a request
            # Because the Modbus server sends continuous values at certain intervals (configurable)
            # Do I need to use: from pymodbus.transaction import ModbusRtuFramer?
            # res = client.read_holding_registers(address=0, count=2, unit=1) # This works but I want it keep reading from the serial buffer because device sends responses continuously
            if not res.isError():
                log.info("Got torque values: Hi = %d, Low = %d", res.registers[0], res.registers[1])  # Successful response :)
            else:
                log.error(res)  # Error response :(
    else:
        # Failed to connect to Modbus Server/Slave device
        log.error('Cannot connect to the Modbus Server/Slave')

if __name__ == '__main__':
    main()
```

dhoomakethu, 2021-03-31:
copying from pymodbus gitter channel

> I am not sure this is possible with pymodbus or in general any modbus compliant client's. There is no streaming option in standard modbus. You will have to read the serial line directly and process it as a modbus frame , take a look at https://github.com/riptideio/pymodbus/blob/master/examples/contrib/message_parser.py to see how you can parse the info.

tracernz, 2021-03-31:
Modbus is probably not the right protocol for this kind of application. EtherCAT or even Ethernet/IP offer the means to do this in a standards compliant way.

closed 2021-05-07 by github-actions[bot]