pymodbus-dev/pymodbus #623: pymodbys

astha172, 2021-04-01:
### Versions

* Python:
* OS:
* Pymodbus:
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

astha172, 2021-04-01:
hi, i connected modbus to pid controller and uses this code to write 

import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer
import time,os

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client= ModbusClient(method = "rtu", port="COM13",stopbits = 1, bytesize = 8, parity = 'N', baudrate=9600)
#connect to the serial modbus server
connection = client.connect()
message='hello'
print(message)
print connection
#Starting add, num of reg to read, slave unit.
#def hello():
while True:
    address=40065
    value=5
    result=client.write_register(0x0041,0x030,unit=0x01)
    print (result)
    time.sleep(2)
        
client.close()

code has output but there is no change in value of pid controller

dhoomakethu, 2021-04-01:
Can you share the logs for the transaction? Also add read holding register
call before and after write.

ಅವರು ಬರೆದಿದ್ದಾರೆ:

astha172, 2021-04-02:
hello
True
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x6 0x0 0xb8 0x0 0xfa 0x89 0xac
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x6 0x0 0xb8 0x0 0xfa 0x89 0xac
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x6 0x0 0xb8 0x0 0xfa
DEBUG:pymodbus.factory:Factory Response[WriteSingleRegisterResponse: 6]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
WriteRegisterResponse 184 => 250

astha172, 2021-04-02:
this is what i m getting ...after writing on 40184......as i seen previously that .....there is to offset of address.....thats why for writting in register185.....i given address of 184

astha172, 2021-04-02:
[130]
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:SEND: 0x1 0x6 0x0 0xb8 0x0 0xfa 0x89 0xac
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1617335505.947966, Current Time stamp - 1617335506.041352
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x6 0x0 0xb8 0x0 0xfa 0x89 0xac
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x6 0x0 0xb8 0x0 0xfa
DEBUG:pymodbus.factory:Factory Response[WriteSingleRegisterResponse: 6]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
WriteRegisterResponse 184 => 250
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 6
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x0 0xb8 0x0 0x1 0x4 0x2f
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - 1617335506.14692, Current Time stamp - 1617335506.212058
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x3 0x2 0x0 0x82 0x38 0x25
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x3 0x2 0x0 0x82
DEBUG:pymodbus.factory:Factory Response[ReadHoldingRegistersResponse: 3]
DEBUG:pymodbus.framer.rtu_framer:Frame advanced, resetting header!!
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
[130]

astha172, 2021-04-02:
after adding read holding register before and after write .....i got same result nothing change value

dhoomakethu, 2021-04-02:
There are no error response received from the slave and it looks like there is no issue with your pymodbus code. Please refer the manual of your slave if it requires some explicit action to enable writing to registers. You can also try with other clients and available and confirm if the behaviour is same or different.

astha172, 2021-04-02:
what can be other different clients
[PID500_Modbus_Doc.pdf](https://github.com/riptideio/pymodbus/files/6247788/PID500_Modbus_Doc.pdf)
this is register list for reading and writing for PID

closed 2021-05-09 by github-actions[bot]