pymodbus-dev/pymodbus #627: Enron Modbus support

mosolovsa, 2021-04-28:
Hi!

Pymodbus doesn't support Enron Modbus ( https://www.simplymodbus.ca/Enron.htm ) extension anyhow, am I right?

And if so, does the PR with such functionality would be eligible?
If the answer is yes I could implement it with some support via high-level guidance about how the author sees it must be done.

In case someone would be curious why the heck do I know that Enron Modbus extension is existing, some industrial measurment manufacturers thinks that it's a good idea to extend existing protocol for the corner cases (e.g. transmitting 4 byte values, e.g. float32 and etc as a single "register").

Krolken, 2021-05-28:
I just got an OMNI gas flow computer in my lab that uses Enron Modbus so I am also interested in implementing the Enron version of Modbus.

swamper123, 2021-05-28:
I would recommened that in case you want to develope such an extension, you may want to start on the 3.0.0 branch (which drops Python2 and so is breaking changes).

dhoomakethu, 2021-05-29:
@Krolken / @mosolovsa Feel free to implement and raise a PR .Also as @swamper123  mentioned please make sure it is python3 compatible.

closed 2021-07-05 by github-actions[bot]

Krolken, 2021-08-04:
Just to follow up. I tried to work out if it was possible to add the Enron modbus in this library. I think some of enron modbus would be OK to implement since it is mostly just supporting 32 bit answers. 
But the real special cases like History readout would just make it really hard to get into the current code. 

So I made a separate python package if anyone is interested. [https://github.com/u9n/enron-modbus](https://github.com/u9n/enron-modbus)