pymodbus-dev/pymodbus #629: Modbus RTU (serial) connection fails after first command in 2.5.1, (works in 2.3.0)

thehaxxa, 2021-05-11:
### Versions

* Python: 3.8.1
* OS: Linux
* Pymodbus: 2.5.1
* Modbus Hardware (if used): 
_MCU via CP2102 (USB)
Rockwell, Siemens PLCs  (FTDI Adapter)
ESP32 over UART (FTDI Adapter)_

### Pymodbus Specific
* Client: rtu - sync

### Description

This issue was identified as a upstream issue with pymodbus from Home Assistant's Modbus Component. 
This issue emerged in Home Assistant 2021.5, when the pymodbus library was updated from 2.3.0 to 2.5.1.

In pymodbus 2.3.0 a Modbus device connected over serial worked but often produced an error on startup in Home Assistant:
`Cleanup recv buffer before send: 0x4d 0xfb` .

After this error occurred the Modbus client would continue to function correctly and the error would not occur again. Communication between the modbus device functioned as expected. 

In pymodbus 2.5.1, serial devices do not work after after the first modbus command. The first command always works but this error is produced and all future modbus commands fail:

```
2021-05-11 02:10:25 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2021-05-11 02:10:25 WARNING (SyncWorker_1) [pymodbus.client.sync] Cleanup recv buffer before send: 0x4d 0xfb
2021-05-11 02:10:25 ERROR (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
```
Reverting back to pymodbus 2.30 fixes the issue. 

It is possible this issue is related: https://github.com/riptideio/pymodbus/issues/510

This is a significant issue, as Modbus RTU does not work after the initial command.
Testing was conducted with USB UART connections and RS485 serial connections to several microcontrollers and PLCs.

Speaking with @janiversen, (_maintainer of Home Assistant Modbus Component_) we identified the issue to be caused by updating to pymodbus 2.5.1.

### Code and Logs

Although configuration applies to Home Assistant, I have included it for reference. NOTE: I am not using pymodbus directly, when testing. 

```yaml
  modbus:
  - name: Curtain Controller
    type: serial
    method: rtu
    port: /dev/serial/by-id/usb-www.freetronics.com_0003_95437313934351505191-if00
    baudrate: 9600
    stopbits: 1
    bytesize: 8
    parity: N
    timeout: 10
    switches:
      #Curtain Controller
      - name: "Bedroom Curtain - UP"
        slave: 1
        address: 0
        write_type: coil
      - name: "Bedroom Curtain - Down"
        slave: 1
        address: 1
        write_type: coil
      - name: "Bedroom Curtain - Stop"
        slave: 1
        address: 2
        write_type: coil

```

janiversen, 2021-05-11:
👍 Lets hope this get solved fast and 2.5.2 are release so we can upgrade.

We want to use pymodbus async version, but once the 3.x is released.

dhoomakethu, 2021-05-11:
@thehaxxa would you mind giving the complete logs with 2.5.1 and 2.3.0. Pymodbus 2.5.1 closes connection on encountering error. We have seen in some cases adding a small delay after an error is encountered fixes these sort of issue. There should also be kwarg which should disable disconnects after an erroneous transaction. Will try to find it.

thehaxxa, 2021-05-11:
> We have the possibility to add a delay between the connect() and the first request, that might help as well.

I can confirm the addition of `delay: 2` and `delay: 5` had no effect in this context.

dhoomakethu, 2021-05-11:
@thehaxxa  No the delay needs to be after an error is received. I will try to see if I can reproduce this issue with the limited hardware I have at my  disposal.

janiversen, 2021-05-11:
@dhoomakethu without being deep into your code, allow me a suggestion:
after connect() in client mode, skip anything received until first request is sent. For server mode, you need to add "If not a valid modbus message".

After that I think disconnect() on bad messages is an excellent idea.

janiversen, 2021-05-11:
If you want that tested I am sure @thehaxxa can help, if you send a patched python file.

dhoomakethu, 2021-05-12:
> @dhoomakethu without being deep into your code, allow me a suggestion:
> after connect() in client mode, skip anything received until first request is sent. For server mode, you need to add "If not a valid modbus message".
> 
> After that I think disconnect() on bad messages is an excellent idea.

We are kind of doing it before every request [here](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/sync.py#L699)

dhoomakethu, 2021-05-13:
@thehaxxa @janiversen Please take a look in to #631 . This is basically a shot in the dark as I was not able to reproduce the error locally with my setup. Let me know if this works. This should be same as the behaviour of the client with pymodbus < 2.5.0.

Let me know if that works. You can patch `transaction.py` locally and see if that helps.

thehaxxa, 2021-05-13:
> @thehaxxa @janiversen Please take a look in to #631 . This is basically a shot in the dark as I was not able to reproduce the error locally with my setup. Let me know if this works. This should be [...]

I tested the commit and made a change to the Modbus Component in Home Assistant to add the `reset_socket=False` parameter. Everyone is working as expected thanks for making this PR.

janiversen, 2021-05-13:
@dhoomakethu first of all a big thanks for your swift response, you saved us for the pain of downgrading. HomeAssistant do however have a policy (which I cannot change) of only using released libraries, so I really really hope you can make a 2.5.2 soon.

I had a bigger look at the test cases, and it seems to me that it would be good to add some "black box" tests, like e.g. write a test that implement a client, mocks the socket so that the mock simulates a device. That way you have a test of the whole stack in one go. If you want to, I can make a couple of PR´s for such an approach.

dhoomakethu, 2021-05-13:
@thehaxxa thanks for the confirmation. @janiversen I do  have plans for functional tests which are long pending.However it's unlikely that is going to be implemented in near future. I will be making a release today or latest by the end of week. Thankd

terminet85, 2021-05-13:
@thehaxxa I've tried the same but I still got crash

you have added reset_socket=false on modbus.py after line 183?

`                self._client = ModbusSerialClient(
                    method=self._config_method,
                    port=self._config_port,
                    baudrate=self._config_baudrate,
                    stopbits=self._config_stopbits,
                    bytesize=self._config_bytesize,
                    parity=self._config_parity,
                    timeout=self._config_timeout,
                    retry_on_empty=True,
                    reset_socket=False,
                )
`

thehaxxa, 2021-05-13:
> @thehaxxa I've tried the same but I still got crash
>
> you have added reset_socket=false on modbus.py after line 183?
>
> ```
>                 self._client = ModbusSerialClient( [...]

That's correct, was it working in 2.3.0? The behaviour is the same for me.

terminet85, 2021-05-13:
@thehaxxa
I'm the guy that was posting on https://github.com/home-assistant/core/pull/50457

> Everything is working as expected! Thanks for making this PR.
Sorry I'm missing something... Is it working for you or not?

thehaxxa, 2021-05-13:
> @thehaxxa
> I'm the guy that was posting on https://github.com/home-assistant/core/pull/50457
> 
> > Everything is working as expected! Thanks for making this PR.
> 
> Sorry I'm missing something... Is it working for you or not?

Yes it is working, my point is this issue was introduced in 2.5 and now it's functional with this parameter.

terminet85, 2021-05-13:
I'm unable to reproduce this fix in my setup. Rock solid on 2.3.0

janiversen, 2021-05-13:
In order to active the fix for modbus you need to change files in the pymodbus library, and change in modbus.py as well.

I expect this patch to help users with instable connections, primarely based on rs-485 (serial).

On dev you will find 2 other user facing patchs and 2 internal patches:
- parameter “timeout:” on “modbus:” can now be used to control slow devices, that is if the device responds after pymodbus reached timeout.
- parameter “delay:” on “modbus:” can now be used to delay sending requests after connect, that is to be used if your device needs time to wakeup after connect (like e.g. huawei sun2000 solar inverters).
Both parameters are valid for all types of connection serial/tcp/udp

The bigger problem is that the modbus integration blocks threads, when a device show signs of trouble, this leads to lovelace become non-responsive. This problem have been present since the start of the year, but have because more “visible” with pymodbus 2.5.2 (simply because it has better error handling). I expect to be able to submit a PR at the end of the week to solve this theme, as you can imagine it is not easy to test, so it takes time to prepare.

I highly expect that with 2021.6 we have a stable modbus integration, and less error prune than with pymodbus 2.3.0, I like the expression often used “2.30 is rock solid”, sounds good, but often when you examine it with debug logs, it simply does not report the errors.

dhoomakethu, 2021-05-13:
@thehaxxa @terminet85 @janiversen Please let me know if I should wait for you guys to confirm the solution to make a release. If you think its not solving the issue, please provide me with more logs/info to see if there is something else I need to take care of.

terminet85, 2021-05-13:
@janiversen yeah i've patched the pymodbus library...

@dhoomakethu If I've forgot something is still no working. I'll take some debug home-assistant log if is fine.

janiversen, 2021-05-13:
OhhI thought I wrote that, I will upgrade pymodbus as soon as 2.5.2 is available.

janiversen, 2021-05-13:
> @janiversen yeah i've patched the pymodbus library...
> 
> @dhoomakethu If I've forgot something is still no working. I'll take some debug home-assistant log if is fine.

Are you sure your issue is the same as @thehaxxa do you use a serial line. You might be better off to try the other options provided on dev.

janiversen, 2021-05-13:
Let us keep the HA issues on HA and if there are pymodbus issuesI will pass them on. @thehaxxa made this issue in accordance with me.

terminet85, 2021-05-13:
> Are you sure your issue is the same as @thehaxxa do you use a serial line. You might be better off to try the other options provided on dev.

Sure... I'm on serial too.
What is the other options please?

janiversen, 2021-05-13:
Delay/timeout on modbus: available on dev.

terminet85, 2021-05-13:
@janiversen I've git cloned modbus from dev, I need to set something? 'cause still crash.

janiversen, 2021-05-13:
Please you need to read what I write on the different issues.
1) Let us not discuss HA problems here, this Issue for a specific bug that have been solved
2) The message before your tells you what you need
3) I am working on a patch that solves the lovelace issue.
But as far as I know (I apologize if I am wrong) you have not supplied logs to show the HA crash, which is interesting because other experience a non-responsive lovelace.

dhoomakethu, 2021-05-13:
Please note, there will be an official 2.5.2 release in sometime.

thehaxxa, 2021-05-13:
> @janiversen I've git cloned modbus from dev, I need to set something? 'cause still crash.

I would open an issue on the HASS github. Providing as much detail of the hardware as you can, turn on debug logging and post your findings there.

terminet85, 2021-05-13:
> * Let us not discuss HA problems here, this Issue for a specific bug that have been solved
> 
> *

If you keep closing thread and tell what not to do, will be hard for me to be helpful. I need to open a new issue in home-assistant so?

janiversen, 2021-05-13:
Either open an issue in HA, or use one of the issues already open to provide your details. But of course if you are sure you have a pymodbus issue (which I as codeowner am not) you are in the right place.

dhoomakethu, 2021-05-13:
This is officially released.Please refer https://github.com/riptideio/pymodbus/releases/tag/v2.5.2 . You can install the latest with `pip install pymodbus --upgrade` or `pip install pymodbus==2.5.2`.

PR #631 (2021-05-13): #629 Add kwarg 'reset_socket' to disable closing sockets on error
Closes #629 

Starting pymodbus 2.5.0 on error client is closed and a new socket is initialized on a new request. This seems to be having adverse effect on the functionality of pymodbus with some of the hardwares. As a workaround, introducing new kwarg `reset_socket` which is set to `True` by default implying socket will be closed on error. Setting this to `False` will keep the socket open till client is explicitly closed or application is stopped.

Example usage
```
client = ModbusSerialClient(method='rtu',....., reset_socket=F [... cut]
closed 2021-05-13 by dhoomakethu