pymodbus-dev/pymodbus #630: serial port speed related log records on every sync transaction since 2.5.0.rc1

dcneeme, 2021-05-12:
Found that when using sync queries, every query creates the following line into /var/log/syslog and /var/log/kern.log. 

```
May 12 07:49:32 localhost kernel: [ 3074.666453@0] meson_uart c81004e0.serial: ttyS4 use xtal(8M) 24000000 change 19200 to 19200
May 12 07:49:33 localhost kernel: [ 3075.688603@0] meson_uart c81004e0.serial: ttyS4 use xtal(8M) 24000000 change 19200 to 19200
May 12 07:49:34 localhost kernel: [ 3076.706737@0] meson_uart c81004e0.serial: ttyS4 use xtal(8M) 24000000 change 19200 to 19200
```

The same information is visible with dmesg -w. 

This is a problem because the higher the communication rate, the faster grow the log files. There is no such problem with pymodbus 2.4.0 or earlier versions. 

The problem only seems to be related to sync transactions, as async with tornado 4.5.2 does not create such log records - at least with pymodbus 2.5.0.rc2

using python 3.7.5 
Linux Khadas 4.9.206 #13 SMP PREEMPT Tue Dec 31 00:37:47 CST 2019 aarch64 aarch64 aarch64 GNU/Linux

dhoomakethu, 2021-05-12:
@dcneeme can you try the stable version `2.5.1` and I believe it should be fixed in that version. It is basically due the fact in 2.5.0 connection would close on every transaction and re-opened .

dcneeme, 2021-05-12:
Thanks, 2.5.1 fixed the sissue.
    N
    
    
    On 12/05/2021 11:16, dhoomakethu wrote:
    
    
      
      @dcneeme
        can you try the stable version 2.5.1 and I believe
        it should be fixed in that version. It is basically due the fact
        in 2.5.0 connection would close on every transaction and
        re-opened .
      —
      [
{
"target": "https://github.com/riptideio/pymodbus/issues/630#issuecomment-839567967",
"url": "https://github.com/riptideio/pymodbus/issues/630#issuecomment-839567967",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2021-05-12 by dhoomakethu

dcneeme, 2021-05-17:
the problem is back, for both 2.5.1 and 2.5.2

installing 2.4.0 clears the erratic behaviour for sync (but disables async with tornado)  :(