pymodbus-dev/pymodbus #634: Decoding Holding Registers to binary format

fess1100, 2021-05-18:
### Versions

* Python: 3.9.1
* OS: Windows 10 Pro 2004
* Pymodbus: 2.5.1
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: rtu

### Description
Need help with decoding Holding Registers to binary format. When reading, I get a value from 8 to 15 bits. I can't read the value from 0 to 7 bits. Thanks

Modbus Slave: ID:1 F:03
0: -47,2 (float32)
1:
2: -7565 (int)
3: -3 (int)
4: 4581 (uint)
5: 5555 (uint)
6: 67 (uint)
7: 76 (uint)
8: 0000 0001 0101 1001 (bool)
9: 0000 0100 0110 0011 (bool)

### Code and Logs

```python
# code
from pymodbus.client.sync import ModbusSerialClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian

client = ModbusSerialClient(method='rtu', port='COM5', baudrate=19200, timeout=3, parity='N', stopbits=1, bytesize=8)

if client.connect():
    res1 = client.read_holding_registers(0, 10, unit=1)
    res2 = client.read_holding_registers(0, 2, unit=1)
    res3 = client.read_holding_registers(2, 1, unit=1)
    res4 = client.read_holding_registers(3, 1, unit=1)
    res5 = client.read_holding_registers(8, 1, unit=1)
    if not res3.isError():
        print(res1)
        print(res1.registers)
        print("-" * 60)
        print(BinaryPayloadDecoder.fromRegisters(res2.registers, Endian.Big, Endian.Big).decode_32bit_float())
        print(BinaryPayloadDecoder.fromRegisters(res3.registers, Endian.Big, Endian.Little).decode_16bit_int())
        print(BinaryPayloadDecoder.fromRegisters(res4.registers, Endian.Big, Endian.Little).decode_16bit_int())
        print(BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_bits())
    else:
        print(res3, 'Error')
        # print('\n')
    print("-" * 60)
else:
    print('Cannot connect Modbus Server/Slave')

# logs
ReadHoldingRegistersResponse (10)
[49724, 52429, 57971, 65533, 4581, 5555, 67, 76, 345, 1123]
------------------------------------------------------------
-47.20000076293945
-7565
-3
[True, False, False, False, False, False, False, False]
------------------------------------------------------------

Process finished with exit code 0

```

dhoomakethu, 2021-05-19:
The`WordOrder` and `ByteOrder` are different for your second decode onwards. May be that is the issue?

fess1100, 2021-05-19:
> The`WordOrder` and `ByteOrder` are different for your second decode onwards. May be that is the issue?

Please specify what you mean. I tried different combinations of register polling

``
#code 1
from pymodbus.client.sync import ModbusSerialClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
from PyQt5 import QtWidgets, uic
from PyQt5.QtSerialPort import QSerialPort, QSerialPortInfo
from PyQt5.QtCore import QIODevice

client = ModbusSerialClient(method='rtu', port='COM5', baudrate=19200, timeout=3, parity='N', stopbits=1, bytesize=8)

if client.connect():
    res1 = client.read_holding_registers(0, 10, unit=1)
    res5 = client.read_holding_registers(8, 1, unit=1)
    if not res1.isError():
        print(res1)
        print(res1.registers)
        print("-" * 60)
        print(BinaryPayloadDecoder.fromRegisters(res5.registers, **Endian.Big, Endian.Big**).decode_bits())
    else:
        print(res1, 'Error')
    print("-" * 60)
else:
    print('Cannot connect Modbus Server/Slave')

#  log 1
ReadHoldingRegistersResponse (10)
[49724, 52429, 57971, 65533, 4581, 5555, 67, 76, 345, 1123]
------------------------------------------------------------
[True, False, False, False, False, False, False, False]
------------------------------------------------------------

#code 2
from pymodbus.client.sync import ModbusSerialClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
from PyQt5 import QtWidgets, uic
from PyQt5.QtSerialPort import QSerialPort, QSerialPortInfo
from PyQt5.QtCore import QIODevice

client = ModbusSerialClient(method='rtu', port='COM5', baudrate=19200, timeout=3, parity='N', stopbits=1, bytesize=8)

if client.connect():
    res1 = client.read_holding_registers(0, 10, unit=1)
    res5 = client.read_holding_registers(8, 1, unit=1)
    if not res1.isError():
        print(res1)
        print(res1.registers)
        print("-" * 60)
        print(BinaryPayloadDecoder.fromRegisters(res5.registers, **Endian.Little, Endian.Big**).decode_bits())
    else:
        print(res1, 'Error')
    print("-" * 60)
else:
    print('Cannot connect Modbus Server/Slave')

#  log 2
ReadHoldingRegistersResponse (10)
[49724, 52429, 57971, 65533, 4581, 5555, 67, 76, 345, 1123]
------------------------------------------------------------
[True, False, False, False, False, False, False, False]
------------------------------------------------------------

#code 3
from pymodbus.client.sync import ModbusSerialClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
from PyQt5 import QtWidgets, uic
from PyQt5.QtSerialPort import QSerialPort, QSerialPortInfo
from PyQt5.QtCore import QIODevice

client = ModbusSerialClient(method='rtu', port='COM5', baudrate=19200, timeout=3, parity='N', stopbits=1, bytesize=8)

if client.connect():
    res1 = client.read_holding_registers(0, 10, unit=1)
    res5 = client.read_holding_registers(8, 1, unit=1)
    if not res1.isError():
        print(res1)
        print(res1.registers)
        print("-" * 60)
        print(BinaryPayloadDecoder.fromRegisters(res5.registers, **Endian.Little, Endian.Little**).decode_bits())
    else:
        print(res1, 'Error')
    print("-" * 60)
else:
    print('Cannot connect Modbus Server/Slave')

#  log 3
ReadHoldingRegistersResponse (10)
[49724, 52429, 57971, 65533, 4581, 5555, 67, 76, 345, 1123]
------------------------------------------------------------
[True, False, False, False, False, False, False, False]
------------------------------------------------------------
``

dhoomakethu, 2021-05-19:
Your question is not clear ` I get a value from 8 to 15 bits. I can't read the value from 0 to 7 bits.` . Are you trying to read values as bytes?. How are you reading values from 8-15bits ?

fess1100, 2021-05-19:
Следующим запросом я хочу считать биты 0-15, но приходят 8-15. Что я делаю не так? В этом и есть мой вопрос

`BinaryPayloadDecoder.fromRegisters(client.read_holding_registers(8, 1, unit=1).registers, Endian.Little, Endian.Little).decode_bits()`

dhoomakethu, 2021-05-20:
A modbus register is 2 bytes long, so you are going to read all 16bits(read 2 bytes) when you read 1 register. If you want to read the next register, you will have to use the next offset. If you want to decode 1 byte at time use`decode_8bit_int` or `decode_8bit_uint` methods from `BinaryPayloadDecoder`. Refer https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus_payload.py#L155

fess1100, 2021-05-21:
> A modbus register is 2 bytes long, so you are going to read all 16bits(read 2 bytes) when you read 1 register. If you want to read the next register, you will have to use the next offset. If you [...]

Thank you for your help! Unfortunately, there is a lack of experience in the field of data distribution. I don't understand how it works, but I solved the problem by looking at the method in the minimalmodbus library. The method seems not elegant, I was counting on a compact request from the pymodbus library

```
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000000001) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000000010) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000000100) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000001000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000010000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000000100000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000001000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000010000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000000100000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000001000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000010000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0000100000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0001000000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0010000000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b0100000000000000) > 0)
print((BinaryPayloadDecoder.fromRegisters(res5.registers, Endian.Big, Endian.Little).decode_16bit_int() & 0b1000000000000000) > 0)
```

```
True
False
False
True
True
False
True
False
True
False
False
False
False
False
False
False
```

fess1100, 2021-05-21:
An elegant solution was found, just about what you said. Offset. I'll share it. Problem solved

```
# First read the register:
res5 = client.read_holding_registers(8, 1, unit=1)
# Now do the decoding, first 8 bits:
bits_in_reg = BinaryPayloadDecoder.fromRegisters(res5,Endian.Big,Endian.Little).decode_bits()
# And 8 bits more, note the += to keep and shift the bits you just read
bits_in_reg += BinaryPayloadDecoder.fromRegisters(res5,Endian.Big,Endian.Little).decode_bits()
```

```
[True, False, False, False, False, False, False, False]
[True, False, False, False, False, False, False, False, True, False, False, False, False, False, False, False]
```

closed 2021-05-22 by dhoomakethu

fess1100, 2021-05-30:
The tests were never successful. Could you tell us how to use a bit mask and a loop to convert an integer of 16 bits into an array [True, False, etc...]? Maybe there is a link