pymodbus-dev/pymodbus #635: Modbus Error: [Input/Output] [Errno 25] Inappropriate ioctl for device

munich-ml, 2021-05-19:
### Versions

* Python: 3.5.3
* OS: Rasbian GNU/Linux 9 (stretch)
* Pymodbus: 2.4.0
* Modbus Hardware (if used): Heidelberg Wallbox Energy Control

### Description
I'm getting a **Modbus Error** once I try to read any input register.

### Code and Logs
```python
from pymodbus.client.sync import ModbusSerialClient

walli = ModbusSerialClient(method="rtu",
                           port="/dev/ttyAMA0",
                           baudrate=19200,
                           stopbits=1,
                           bytesize=8,
                           timeout=0.2,
                           parity="E")

walli.connect()
print(walli.__repr__())

r = walli.read_input_registers(13, 1)
print(r)
```
Produces the following output on the Raspberry Pi:

```
Python 3.5.3 (default, Jan 19 2017, 14:11:04) 
[GCC 6.3.0 20170124] on linux
Type "copyright", "credits" or "license()" for more information.
>>> 
=========== RESTART: /home/pi/Desktop/raspi_walli/modbus_trial.py ===========
<ModbusSerialClient at 0x75bcb8b0 socket=Serial<id=0x7534e9d0, open=True>(port='/dev/ttyAMA0', baudrate=19200, bytesize=8, parity='E', stopbits=1, timeout=0.2, xonxoff=False, rtscts=False, dsrdtr=False), method=rtu, timeout=0.2>

Modbus Error: [Input/Output] [Errno 25] Inappropriate ioctl for device
>>> 
```

fess1100, 2021-05-20:
Try this way

r = walli.read_input_registers(13, 1)
print(r.**_registers_**)

dhoomakethu, 2021-05-20:
@munich-ml This error is generally coming from the underlying `pyserial` stack. You will have to make sure your serial port settings are correct, check system/dmesg logs for more info.

closed 2021-05-20 by dhoomakethu

bilalh96, 2021-11-03:
> Try this way
> 
> r = walli.read_input_registers(13, 1) print(r.**_registers_**)

what I dont understand is there is no attribute called "registers" in the module then how does this solve the problem , if it really does , for me it dosent :(

munich-ml, 2021-11-03:
hi @bilalh96,
in my case it really was a permission thing. This will probably print Error on your case:
```python
r = client.read_input_registers(4, count=15, unit=1)
if r.isError():
    print("Error")
else:
    print(r.registers)
```
If so, see this part of my repo for UART setup on Raspberry Pi: https://github.com/munich-ml/raspi_walli/tree/main/raspi_uart

bilalh96, 2021-11-03:
> hi @bilalh96, in my case it really was a permission thing. This will probably print Error on your case:
>
> ```python
> r = client.read_input_registers(4, count=15, unit=1)
> if r.isError(): [...]

is unit =1 slave/server address of device? It can be diff too right

munich-ml, 2021-11-03:
`unit` is the slave address. In case of my Heidelberger Wallbox Energy Contact, I can program the slave address 1.. 15.

bilalh96, 2021-11-03:
> `unit` is the slave address. In case of my Heidelberger Wallbox Energy Contact, I can program the slave address 1.. 15.

can I have your email where I can discuss my specific issue with this unit id rather than extending this chat here for no reason? thank you :)

munich-ml, 2021-11-03:
I believe there can be value in our discussion for other users, too. So I'd rather continue in here.