pymodbus-dev/pymodbus #638: fin ack

OldTinfoil, 2021-05-27:
### Versions

* Python: 3.5.3
* OS: 
* Pymodbus:
* Modbus Hardware (if used): n/a

### Pymodbus Specific
* Server: tcp - (updating) sync
* Client: (3rd party)

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

### Code and Logs

```python
 # code and logs here.

```

OldTinfoil, 2021-05-27:
Apologies - accidentally hit enter too early. Still editing the issue.

OldTinfoil, 2021-05-27:
There we go. Finished butchering this issue.

dhoomakethu, 2021-05-29:
@OldTinfoil Do you also observe the same with Pymodbus 2.5.2?

closed 2021-07-05 by github-actions[bot]

OldTinfoil, 2021-07-05:
We haven't been able to replicate the issue with the version in use in the lab, nor with 2.5.2. Due to the pandemic, I haven't been able to travel to the remote site to investigate the issue in person. I'll loop back around if I make any headway with the issue.