pymodbus-dev/pymodbus #641: AttributeError: 'ModbusUdpClient' object has no attribute '_in_waiting'

MAKOMO, 2021-06-05:
### Versions

* Python: 3.9.5
* OS: macOS
* Pymodbus: v2.5.2
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: tcp/rtu- sync

### Description

The ModbusTcpClient and ModbusUdpClient with retry_on_empty=True or retry_on_invalid=True fail with

```
AttributeError: 'ModbusUdpClient' object has no attribute '_in_waiting'
````

as `_in_waiting` is only defined for `ModbusSerialClient`.

### Code and Logs

```python
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/client/common.py", line 150, in mask_write_register
    return self.execute(request)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/client/sync.py", line 109, in execute
    return self.transaction.execute(request)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/transaction.py", line 194, in execute
    response, last_exception = self._retry_transaction(retries, "empty", request, expected_response_length, full=full)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pymodbus/transaction.py", line 250, in _retry_transaction
    in_waiting = self.client._in_waiting()
AttributeError: 'ModbusUdpClient' object has no attribute '_in_waiting'
```

georgezhao2010, 2021-06-18:
I got this issue too.

dhoomakethu, 2021-08-01:
Fixed with #659

closed 2021-08-01 by dhoomakethu