pymodbus-dev/pymodbus #642: Proposing a PR to fix a few small typos

timgates42, 2021-06-09:
# Issue Type

[x] Bug (Typo)

# Steps to Replicate and Expected Behaviour

- Examine examples/common/asynchronous_processor.py, examples/contrib/asynchronous_asyncio_modbus_tls_client.py, examples/contrib/concurrent_client.py, examples/contrib/modbus_saver.py, examples/contrib/modbus_tls_client.py and observe `neccessary`, however expect to see `necessary`.
- Examine pymodbus/repl/client/mclient.py and observe `seperated`, however expect to see `separated`.
- Examine examples/contrib/bcd_payload.py, examples/contrib/modicon_payload.py, pymodbus/payload.py and observe `reponse`, however expect to see `response`.
- Examine examples/contrib/modbus_simulator.py, examples/contrib/remote_server_context.py, pymodbus/datastore/context.py and observe `existance`, however expect to see `existence`.
- Examine pymodbus/client/sync.py and observe `continuosly`, however expect to see `continuously`.
- Examine examples/common/async_asyncio_client.py, examples/common/async_asyncio_serial_client.py and observe `ayncio`, however expect to see `asyncio`.
- Examine pymodbus/device.py and observe `singleotn`, however expect to see `singleton`.
- Examine README.rst and observe `respository`, however expect to see `repository`.
- Examine README.rst and observe `recomended`, however expect to see `recommended`.
- Examine pymodbus/utilities.py and observe `recieved`, however expect to see `received`.
- Examine examples/common/asynchronous_processor.py and observe `procesing`, however expect to see `processing`.
- Examine examples/gui/bottle/media/js/backbone.js and observe `prefered`, however expect to see `preferred`.
- Examine examples/tools/build_datastore.py and observe `pickeld`, however expect to see `pickled`.
- Examine pymodbus/constants.py and observe `optinoal`, however expect to see `optional`.
- Examine examples/contrib/message_parser.py and observe `messge`, however expect to see `message`.
- Examine pymodbus/datastore/store.py and observe `intially`, however expect to see `initially`.
- Examine examples/gui/bottle/frontend.py and observe `initialied`, however expect to see `initialized`.
- Examine test/test_server_sync.py and observe `intial`, however expect to see `initial`.

# Notes

Semi-automated issue generated by
https://github.com/timgates42/meticulous/blob/master/docs/NOTE.md

To avoid wasting CI processing resources a branch with the fix has been
prepared but a pull request has not yet been created. A pull request fixing
the issue can be prepared from the link below, feel free to create it or
request @timgates42 create the PR. Alternatively if the fix is undesired please
close the issue with a small comment about the reasoning.

https://github.com/timgates42/pymodbus/pull/new/bugfix_typos

Thanks.

dhoomakethu, 2021-06-10:
Thanks @timgates42  Please raise a PR against the `dev` branch.

timgates42, 2021-07-01:
Thanks @dhoomakethu raised #649 against the `dev` branch

closed 2021-08-01 by dhoomakethu