pymodbus-dev/pymodbus #644: ModbusTlsServer should use ModbusTlsFramer by default, NOT ModbusSocketFramer.

aneesahmedpro, 2021-06-14:
### Versions

* Pymodbus: 2.5.2

### Pymodbus Specific
* Server: tls - sync
* Client: none

### Description

(Technically, this a feature request, but trust me it feels like a bug.)

I was trying to create a TLS modbus server. Since I wanted to start and stop the server manually, I chose to use `pymodbus.server.sync.ModbusTlsServer` instead of `pymodbus.server.sync.StartTlsServer`. But it did not work, I was only getting `Frame check failed, ignoring!!` messages in debug log.

After snooping around in the source code, I found out that the constructor of `ModbusTlsServer` does not create a `ModbusTlsFramer` to pass as the framer to the base class (`ModbusTcpServer`). So, if we do not specify a framer explicitly, `ModbusTlsServer` ends up using `ModbusSocketFramer`, which is incompatible.

Kindly change this behaviour, as it wastes time of and causes confusion for any developer who is new to pymodbus library. As the developer would not really know that framers are a thing and that wrong one is being used.

I see that `pymodbus.server.sync.StartTlsServer` does handle this automatically, but it only helps the developers who are not starting and stopping the server manually.

----

P.S. Kudos to all developers for creating this awesome library :-D

starnight, 2021-07-25:
Thanks for your report. :)
We can make it more robust here.  I am going to have a PR to enhance it.

janiversen, 2022-05-25:
Solved.

commit 406683d5 (2021-08-01): sync/ModbusTlsServer: Fix ModbusTlsServer's default framer

ModbusTlsServer's framer is set as ModbusTlsFramer by caller
StartTlsServer. However, if developers invoke ModbusTlsServer directly
without assigned framer, the framer will be assigned as
ModbusSocketFramer. Because, ModbusTlsServer inherits from
ModbusTcpServer. This parses payload failed for ModbusTlsServer.
This patch assigns ModbusTlsServer's default framer as ModbusTlsFramer.
And, adds corresponding unittest. (Add sync/ModbusTlsClient's framer
test as well)

Fixes #644
closed 2022-05-25 by janiversen