pymodbus-dev/pymodbus #646: Slaves start not responding when multiple ones are running on the bus

ovunccetin, 2021-06-20:
### Versions

* Python: 3.7.3
* OS: Raspbian GNU/Linux 10 (buster)
* Pymodbus: 2.5.2
* Modbus Hardware (if used): Raspberry Pi Zero W Rev 1.1

### Pymodbus Specific
* Server: rtu - sync/async
* Client: rtu - sync

### Description

I have two identical slaves (RPi Zero) and a master (RPi 4) on the Modbus. Each slave runs a sync RTU server listening port '/dev/serial0'. The slave code is the modified version of the Pymodbus' synchronous server example.

In the master, I have two separate client processes each of which periodically hits a separate slave. In other words, client process C1 reads from slave S1, whereas C2 reads from S2 every second (i.e. 1 RPS).

In this setup, the slaves cannot properly respond. Both slaves start to slow down and I'm seeing "No response" (_Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response_) errors on clients. When I stop one of the slaves, the other slave and its client returns to normal and the response errors disappear. When I start the stopped slave again, the problem reoccurs. Moreover, in the single slave scenario, the running slave can respond higher number of requests like 10 or 20 RPS.

I have also tried the async server (Twisted) on slave side. The problem still occurs. 
 
### Code and Logs

Client Code (sync):
```python
UNIT_ID = 1
TIMEOUT = 0.005
BAUDRATE = 9600

def run_server():
    store = ModbusSlaveContext(
        di=ModbusSequentialDataBlock(0, [17]*100),
        co=ModbusSequentialDataBlock(0, [17]*100),
        hr=ModbusSequentialDataBlock(0, [17]*100),
        ir=ModbusSequentialDataBlock(0, [17]*100))

    context = ModbusServerContext(slaves={UNIT_ID: store}, single=False)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #
    # If you don't set this or any fields, they are defaulted to empty strings.
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'Pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Pymodbus Server'
    identity.ModelName = 'Pymodbus Server'
    identity.MajorMinorRevision = version.short()

    # RTU:
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity,
                      port='/dev/serial0', timeout=TIMEOUT, baudrate=BAUDRATE,
                      ignore_missing_slaves=IGNORE_MISSING_SLAVES)

if __name__ == "__main__":
    run_server()
```

Thanks,
Ovunc

dhoomakethu, 2021-06-28:
Are both slaves connected on the same port ?

janiversen, 2022-08-27:
You can only have one client pr physical connection. When you switch client will sometimes loose part of the traffic.

When both clients keep the same port open, you have problems when sending requests.

closed 2022-08-27 by janiversen