pymodbus-dev/pymodbus #648: Pymodbus: After StopServer -> StartSerialServer doesn't work (could not open port: PermissionError)

tiberryb, 2021-07-01:
### **Versions**

* Python: 3.9
* OS: Raspian
* Pymodbus: 2.5.1
* Modbus Hardware (if used): Raspberry Pi 3B

### **Pymodbus Specific**
* Server: rtu - sync/async
* Client: rtu - sync/async

### **Description**

In my application I want to start a SerialServer, which can be read by a client connected over RTU. After the client has read the values I want to stop the server. At this point the user should be able to change some values himself and start the SerialServer again. Exactly the second start of the SerialServer doesn't work, because the server can't open the port.

### **Code and Logs**

```python
# Code:
from pymodbus.version import version
from pymodbus.server.asynchronous import StartSerialServer, StopServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer

import time
import sys
import threading
from qtpy import QtWidgets

app = QtWidgets.QApplication(sys.argv)

def modbus_server_start():
    print("Modbus-Server start:")

    class CustomDataBlock(ModbusSparseDataBlock):
        def setValues(self, address, value):
            super(CustomDataBlock, self).setValues(0, value_1)
            super(CustomDataBlock, self).setValues(1, 11)
            super(CustomDataBlock, self).setValues(2, 12)

    block = CustomDataBlock({0: [0] * 100})
    store = ModbusSlaveContext(hr=block)
    context = ModbusServerContext(slaves=store)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'pymodbus Server'
    identity.ModelName = 'pymodbus Server'
    identity.MajorMinorRevision = version.short()

    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='COM6', timeout=0.05, baudrate=19200)

def server_start():
    print("Start an async server.")
    t = threading.Thread(target=modbus_server_start)
    t.daemon = True
    t.start()

def modbus_server_stop():
    stopthread = threading.Thread(target=stop)
    stopthread.daemon = True
    stopthread.start()
    print("Server was stopped.")

def stop():
    print("Process will be down.")
    StopServer()  # Stop server.
    sys.exit(0)  # Kill the server code.

window = QtWidgets.QMainWindow()
window.setWindowTitle("StartSerialServer")
window.setGeometry(500, 500, 400, 125)

button = QtWidgets.QPushButton(window)
button.setText("Start")
button.setGeometry(0, 0, 60, 30)
button.show()
button.clicked.connect(server_start)

button2 = QtWidgets.QPushButton(window)
button2.setText("Stop")
button2.setGeometry(100, 0, 60, 30)
button2.show()
button2.clicked.connect(modbus_server_stop)

combobox = QtWidgets.QComboBox(window)
combobox.setGeometry(50, 40, 60, 30)
combobox.addItem("")
combobox.addItem("")
combobox.addItem("")
combobox.addItem("")
combobox.addItem("")
combobox.setItemText(0, "0")
combobox.setItemText(1, "10")
combobox.setItemText(2, "20")
combobox.setItemText(3, "30")
combobox.setItemText(4, "40")
combobox.show()
global value_1
value_1 = combobox.currentText()

window.show()
sys.exit(app.exec_())

# Logs:
Start an async server.
Modbus-Server start:
Process will be down.
Server was stopped.

Exception in thread Thread-4:
Traceback (most recent call last):
  File "C:\Python32\lib\threading.py", line 954, in _bootstrap_inner
Start an async server.
Modbus-Server start:
    self.run()
  File "C:\Python32\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python32\Projekte\Pymodbus-Test\github_issue.py", line 36, in modbus_server_start
    StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='COM6', timeout=0.05, baudrate=19200)
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\pymodbus\server\asynchronous.py", line 340, in StartSerialServer
    SerialPort(protocol, port, reactor, baudrate=baudrate, parity=parity,
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\twisted\internet\_win32serialport.py", line 44, in __init__
    self._serial = self._serialFactory(
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM6': PermissionError(13, 'Zugriff verweigert', None, 5)
```
I thought about closing and reopen the server but this also doesn't work.

Maybe somebody could help me to fix the problem. :)

dhoomakethu, 2021-07-01:
Looks like some part of your code is keeping the port occupied, see if there are any stray processes/threads running in the background which are not killed.

tiberryb, 2021-07-12:
I also tried sync server but there was the same issue. I think the problem is, when I want to close and stop the server the server_close function wants to reconnect to the client but this is not possible because when I start the server the Client in already connected. Is it possible to stop the server without connencting the client again?

```
DEBUG:pymodbus.server.sync:Client Connected [COM6:COM6]
DEBUG:pymodbus.server.sync:Modbus server stopped
DEBUG:pymodbus.server.sync:Client Disconnected [COM6:COM6]
```

dhoomakethu, 2021-07-15:
@tiberryb the client connects to server and not the other way around I believe. When the server is stopped , the server will disconnect all active connections.

tiberryb, 2021-07-16:
@dhoomakethu thanks for your reply. 
But there is the problem. When I want to stop the server after starting it, this doesn't work. The server gets no permission because when the server is stopped, he wants to connect to the client first. In my application I only want to disconnect from the client without reconnecting.

**Log:**

**Server was started.**
```
DEBUG:pymodbus.server.sync:Client Connected [COM6:COM6]
DEBUG:pymodbus.server.sync:Started thread to serve client
```
**Server was stopped.**
```
ERROR:pymodbus.server.sync:could not open port 'COM6': PermissionError(13, 'Zugriff verweigert', None, 5)
DEBUG:pymodbus.server.sync:Modbus server stopped
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python32\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "C:\Python32\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Python32\Projekte\Pymodbus-Test\main_kombi.py", line 301, in stop
    StopSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='COM6', timeout=0.05, baudrate=19200)
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\pymodbus\server\sync.py", line 702, in StopSerialServer
    server.server_close()
  File "C:\Python32\Projekte\Pymodbus-Test\venv\lib\site-packages\pymodbus\server\sync.py", line 598, in server_close
    self.handler.finish()
AttributeError: 'NoneType' object has no attribute 'finish'

```

dhoomakethu, 2021-07-21:
That sound OS specific issue and nothing to do with pymodbus. It looks likes some of the resources are not released when you stop the server, it could be that there is an active connection in place. You can start with stopping and restarting server with out a client connected and repeat the experiment with a client connected.

closed 2021-08-24 by tiberryb