pymodbus-dev/pymodbus #654: Help required with connection to sensor

ArjunAnantharaman, 2021-07-27:
### Versions

* Python: 3.8
* OS: Windows
* Pymodbus: 2.5.2

### Description

I am new to Pymodbus and I want to establish a connection to a sensor that seems to use the half-duplex protocol. From my reading and searching online, I used pymodbus with multiple implementations, the most descriptive being with the ModBusClient. I have a sensor whose data is publically available at https://www.micro-epsilon.de/download/manuals/man--inertialSENSOR-INC5701--en.pdf. 

In the last two pages, one can see the examples required to request data from the slave and I tried a similar implementation with the client_read_holding_registers, but do not receive any response. I am not sure how to format the "count" and would appreciate any help with this. The error is as follows:

'
result= client.read_input_registers(0x01,count=4,unit=0x7E)
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:SEND: 0x7e 0x4 0x0 0x1 0x0 0x4 0xab 0xc6
DEBUG:pymodbus.framer.rtu_framer:Changing state to IDLE - Last Frame End - None, Current Time stamp - 1627399460.889268
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:No response received, Expected 13 bytes Recieved 0 bytes !!!!
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 
DEBUG:pymodbus.framer.rtu_framer:Frame - [b''] not ready
DEBUG:pymodbus.transaction:Getting transaction 126
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
'

When printing the results.registers, I get the error 'ModbusIOException' object has no attribute 'registers'

In the end, I want to read data continuously from this sensor and store it to a file every day or so as a new file. 

### Code and Logs

```python
import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#count= the number of registers to read
#unit= the slave unit this request is targeting
#address= the starting address to read from

client= ModbusClient(method = "rtu", port="COM10", stopbits = 1, bytesize = 8,
                     parity = 'E', baudrate= 230400)

#Connect to the serial modbus server
connection = client.connect()
print (connection)

#Starting add, num of reg to read, slave unit.
result= client.read_input_registers(0x01,count=4,unit=126)

print(result.registers)
```

ScrabbleProgrammer, 2021-07-28:
You're using parity 'E'. Try changing it to 'N'. Also, make sure the baudrate is the same as on the sensor.

ArjunAnantharaman, 2021-07-28:
> You're using parity 'E'. Try changing it to 'N'. Also, make sure the baudrate is the same as on the sensor.

I tried with Parity 'N', still the same result, and the baudrate is as specified by the sensor manual. I even tried changing the baud rate in the device manager in windows to match that of the sensor (9600 is the default for those COM ports) but the same error persists.

dhoomakethu, 2021-07-28:
If you are using on windows, try `strict=False` while initialising the client (`ModbusClient(method=..., strict=False)`).

dhoomakethu, 2021-07-28:
In any case, the documentation does not specify the protocol to be `modbus` but just plain RS485. I do not think pymodbus can help you there. Try `pyserial` library directly to read the data with the frame structure mentioned in the document,

closed 2021-07-28 by dhoomakethu

ArjunAnantharaman, 2021-07-28:
> In any case, the documentation does not specify the protocol to be `modbus` but just plain RS485. I do not think pymodbus can help you there. Try `pyserial` library directly to read the data with the frame structure mentioned in the document,

I tried to use the strict, but with the same results. I was initially trying only with pyserial to directly use the ser.readline() command to get the data from the sensor. When the port opened (similar to this), there was no data being sent to be read. 

Again, this could be a very basic mistake on my part since I have not worked with this before, so thank again, and I hope further suggestions can help me sort this out :)