pymodbus-dev/pymodbus #655: ImportError on Raspberry Pi 4 terminal

ScrabbleProgrammer, 2021-07-28:
### Versions

* Python: 3.9.6
* OS: Raspbian OS
* Pymodbus: 2.5.2

### Description

Hello, I'm currently using PyModbus to communicate with a number of sensors via RS485 from a Raspberry Pi 4 and occasionally my PC. And although the script runs perfectly from an IDE, I'm unable to run the script through the terminal. For some reason, the Raspberry Pi 4 is unable to import the PyModbus module, even though the built-in IDE (Thonny Python IDE) can. I've used the same script through my PC and it works on the terminal but the same cannot be said with the Raspberry Pi 4. I'd appreciate any help. 

Error logs are written below. Note that I've installed Pymodbus via pip in the Raspberry Pi 4.

### Code and Logs

```
pi@raspberrypi:~/Desktop $ python modbus_monitoring.py
Traceback (most recent call last):
 File "modbus_monitoring.py", line 7, in <module>
  from pymodbus.client.sync import ModbusSerialClient as ModbusClient
ImportError: No module named pymodbus.client.sync

```

mindforger, 2021-07-28:
make sure your "default" python is 3.9.6 in your terminal, just run the python console and check there, because as far as i am aware default python is still python2 on raspbian

try /python3 modbus_monitoring.py isntead to selectively call python3 instead of 2 for testing

dhoomakethu, 2021-07-28:
This is not an issue with pymodbus per say, Please check your python path and make sure you are using the correct python interpreter.

closed 2021-07-28 by dhoomakethu