pymodbus-dev/pymodbus #656: Correct way to keep connection open?

track02, 2021-07-28:
### Versions

* Python: 3.8
* OS: MacOS / Linux
* Pymodbus: 2.5
* Modbus Hardware (if used): N/A

### Pymodbus Specific
* Client: tcp async

### Description

Using Pymodbus as a client along with a Windows modbus simulator (Modbus Slave)

I've been doing some testing today, creating a client doing a few reads and writes but if I leave the program running and come back after a few minutes and make another I get an error and my client looks to have been nulled out

Completely new to modbus so apologies if I'm just missing something obvious but is there a proper way to keep a connection 'open' or do I need to check prior to each read/write and recreate it if it's no longer there?

### Code and Logs

I've modified the async example slightly to recreate the issue, it'll write/read a coil and then sit and wait for 5 minutes before making another read and write. 

Tried it a couple of times and the error looks to be consistent

```python
from pymodbus.client.asynchronous.tcp import AsyncModbusTCPClient as ModbusClient
from pymodbus.client.asynchronous import schedulers
import asyncio

SLAVE_ID = 1
SLAVE_ADDRESS = "192.168.10.34"
SLAVE_PORT = 5503

async def start_async_test(client):

    print("-----Writing / Reading Coils-----")  # Address, Coil Count, Slave ID
    wr = await client.write_coil(0, True, unit=SLAVE_ID)  # Single Coil Write
    print(f"S.Write Response Code: {wr.function_code}")
    rr = await client.read_coils(0, 1, unit=SLAVE_ID)
    print(f"Read Response Code: {rr.function_code}")
    print(f"Read Result: {rr.bits}")

    print("Passing some time")
    await asyncio.sleep(300)
    print("Time passed")

    print("-----Writing / Reading Coils-----")  # Address, Coil Count, Slave ID
    wr = await client.write_coil(0, False, unit=SLAVE_ID)  # Single Coil Write
    print(f"S.Write Response Code: {wr.function_code}")
    rr = await client.read_coils(0, 1, unit=SLAVE_ID)
    print(f"Read Response Code: {rr.function_code}")
    print(f"Read Result: {rr.bits}")

def run_with_no_loop():
    loop, client = ModbusClient(schedulers.ASYNC_IO, host=SLAVE_ADDRESS, port=SLAVE_PORT)
    loop.run_until_complete(start_async_test(client.protocol))
    loop.close()

if __name__ == '__main__':

    # Run with no active loop
    run_with_no_loop()
```

Program output: 

```
-----Writing / Reading Coils-----
S.Write Response Code: 5
Read Response Code: 1
Read Result: [True, False, False, False, False, False, False, False]
Passing some time
Time passed
-----Writing / Reading Coils-----
Traceback (most recent call last):
  File "/Users/alex/PycharmProjects/TestModbus/main.py", line 43, in <module>
    run_with_no_loop()
  File "/Users/alex/PycharmProjects/TestModbus/main.py", line 36, in run_with_no_loop
    loop.run_until_complete(start_async_test(client.protocol))
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/Users/alex/PycharmProjects/TestModbus/main.py", line 25, in start_async_test
    wr = await client.write_coil(0, False, unit=SLAVE_ID)  # Single Coil Write
  File "/Users/alex/PycharmProjects/TestModbus/venv/lib/python3.8/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 34, in execute
    req = self._execute(request)
  File "/Users/alex/PycharmProjects/TestModbus/venv/lib/python3.8/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 141, in _execute
    self.write_transport(packet)
  File "/Users/alex/PycharmProjects/TestModbus/venv/lib/python3.8/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 131, in write_transport
    return self.transport.write(packet)
AttributeError: 'NoneType' object has no attribute 'write'
```

dhoomakethu, 2021-07-28:
what particular version of pymodbus is this ? the latest is 2.5.2

track02, 2021-07-28:
@dhoomakethu - Apologies, this is 2.5.2

dhoomakethu, 2021-07-31:
Is slave a real device or a simulated modbus slave ?

dhoomakethu, 2021-07-31:
Also, please enable the debug logs to get more info on what;s happening with the client internally.
You can add the below code block to your code.

```
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)
```

track02, 2021-08-03:
I'll have access to the simulator later this week, will enable the logs and re-run 

This is the slave simulator https://modbustools.com/modbus_slave.html being used, happy to try another for comparison if you have any recommendations

track02, 2021-08-05:
Couldn't see any detailed logs available in the simulator apart from a sparse traffic log which I assume matches up with the first write and read:

```
Rx:000000-11:18:52.226-00 01 00 00 00 06 01 05 00 00 FF 00 
Tx:000001-11:18:52.226-00 01 00 00 00 06 01 05 00 00 FF 00 
Rx:000002-11:18:52.258-00 02 00 00 00 06 01 01 00 00 00 01 
Tx:000003-11:18:52.258-00 02 00 00 00 04 01 01 01 01 
```

Only had a Python 3.7 machine this time, but same error - enabling logs as suggested gives the following:

Seems like it is disconnecting and then reconnecting

```
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:pymodbus.client.asynchronous.async_io:Connecting to 192.168.10.14:11502.
DEBUG:pymodbus.client.asynchronous.async_io:Connecting.
-----Writing / Reading Coils-----
DEBUG:pymodbus.client.asynchronous.async_io:Client connected to modbus server
INFO:pymodbus.client.asynchronous.async_io:Protocol made connection.
INFO:pymodbus.client.asynchronous.async_io:Connected to 192.168.10.14:11502.
DEBUG:pymodbus.client.asynchronous.async_io:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x5 0x0 0x0 0xff 0x0
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.client.asynchronous.async_io:recv: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x5 0x0 0x0 0xff 0x0
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x5 0x0 0x0 0xff 0x0
DEBUG:pymodbus.factory:Factory Response[WriteSingleCoilResponse: 5]
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.client.asynchronous.async_io:send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x0 0x0 0x1
DEBUG:pymodbus.transaction:Adding transaction 2
S.Write Response Code: 5
Read Response Code: 1
Read Result: [True, False, False, False, False, False, False, False]
Passing some time
DEBUG:pymodbus.client.asynchronous.async_io:recv: 0x0 0x2 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x2 0x0 0x0 0x0 0x4 0x1 0x1 0x1 0x1
DEBUG:pymodbus.factory:Factory Response[ReadCoilsResponse: 1]
DEBUG:pymodbus.transaction:Getting transaction 2
DEBUG:pymodbus.client.asynchronous.async_io:Client disconnected from modbus server: None
INFO:pymodbus.client.asynchronous.async_io:Protocol lost connection.
DEBUG:pymodbus.client.asynchronous.async_io:Waiting 100 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.async_io:Connecting.
DEBUG:pymodbus.client.asynchronous.async_io:Client connected to modbus server
INFO:pymodbus.client.asynchronous.async_io:Protocol made connection.
INFO:pymodbus.client.asynchronous.async_io:Connected to 192.168.10.14:11502.
DEBUG:pymodbus.client.asynchronous.async_io:Client disconnected from modbus server: None
INFO:pymodbus.client.asynchronous.async_io:Protocol lost connection.
DEBUG:pymodbus.client.asynchronous.async_io:Waiting 100 ms before next connection attempt.
DEBUG:pymodbus.client.asynchronous.async_io:Connecting.
DEBUG:pymodbus.client.asynchronous.async_io:Client connected to modbus server
INFO:pymodbus.client.asynchronous.async_io:Protocol made connection.
INFO:pymodbus.client.asynchronous.async_io:Connected to 192.168.10.14:11502.
Time passed
-----Writing / Reading Coils-----
DEBUG:pymodbus.client.asynchronous.async_io:send: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x5 0x0 0x0 0x0 0x0
Traceback (most recent call last):
  File "/home/alex/PycharmProjects/Modbus_Scratch/modbus_test.py", line 44, in <module>
    run_with_no_loop()
  File "/home/alex/PycharmProjects/Modbus_Scratch/modbus_test.py", line 38, in run_with_no_loop
    loop.run_until_complete(start_async_test(client.protocol))
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "/home/alex/PycharmProjects/Modbus_Scratch/modbus_test.py", line 29, in start_async_test
    wr = await client.write_coil(0, False, unit=SLAVE_ID)  # Single Coil Write
  File "/home/alex/PycharmProjects/Modbus_Scratch/venv/lib/python3.7/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 34, in execute
    req = self._execute(request)
  File "/home/alex/PycharmProjects/Modbus_Scratch/venv/lib/python3.7/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 141, in _execute
    self.write_transport(packet)
  File "/home/alex/PycharmProjects/Modbus_Scratch/venv/lib/python3.7/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 131, in write_transport
    return self.transport.write(packet)
AttributeError: 'NoneType' object has no attribute 'write'
```

janiversen, 2022-08-29:
Tested ModbusTcpClient and ModbusAsyncTcpClient with a server (no real device) and had it do a couple of requests, sleep for 10 hours and then repeat. So far it have been working for more than 2 days.

Be aware that many devices have a connection timeout when there are no requests. This is a typical implementation to automate a connection reset in case of network problems. If you want standing a standing connection, please make sure you do e.g. a read_coil every now and then.

closed 2022-08-29 by janiversen