pymodbus-dev/pymodbus #657: ModbusTCP _inWaiting not defined.

janiversen, 2021-07-31:
### Versions

* Python: 3.8
* OS: hastío
* Pymodbus: 1.51
* Modbus Hardware (if used): 

### Pymodbus Specific
* Client: tcp - sync

### Description
The modbus integration in home assistant uses pymodbus, and lately we have started getting more and more issues with
```
File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 305, in _pymodbus_call
result = self._pb_call[use_call][ENTRY_FUNC](address, value, **kwargs)
File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 103, in write_registers
return self.execute(request)
File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 109, in execute
return self.transaction.execute(request)
File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 194, in execute
response, last_exception = self._retry_transaction(retries, "empty", request, expected_response_length, full=full)
File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 250, in _retry_transaction
in_waiting = self.client._in_waiting()
AttributeError: 'ModbusTcpClient' object has no attribute '_in_waiting'
```

We use slave 0 as default, and some users inform that the error goes away if they specify a slave > 0, but this is not consistent.

I am the maintainer of the modbus integration.

### Code and Logs

```python
 # code and logs here.
https://github.com/home-assistant/core/tree/dev/homeassistant/components/modbus
```

dhoomakethu, 2021-07-31:
Is the issue also seen with 2.5.2?

dhoomakethu, 2021-07-31:
Never mind this is a bug.

janiversen, 2021-07-31:
Yup but difficult to reproduce and isolate, I have tried for a while….but of course you know the code a lot better.

Looking forward to see a fix.

dhoomakethu, 2021-08-09:
Please give a try with [2.5.3rc1](https://github.com/riptideio/pymodbus/releases/tag/v2.5.3rc1) .
PyPi: https://pypi.org/project/pymodbus/2.5.3rc1/

janiversen, 2021-08-09:
will do

janiversen, 2021-08-09:
As far as I can tell it solves the issue, however I can only do limited testing. I have issued a PR to have 2.5.3rc1 on our dev, allowing much more users to test.

Hvitnov, 2021-09-24:
After update to 2.5.1 I have seen this issue pretty consistently on several dataloggers communicating on modbusTCP via Narrowband IOT.

Updating to 2.5.3rc1 solved this for me.

PR #659 (2021-08-01): Closes #657
Close #657 
Use `in_waiting` only with serial clients.
commit 52037bf4 (2021-08-01): Fix #657 (#659) Use _in_waiting only for serial clients
closed 2021-09-24 by janiversen