pymodbus-dev/pymodbus #667: pymodbus tcp client does not clean socket when not receiving data.

janiversen, 2021-09-14:
### Versions

* Python:
3.8.2
* OS:
hassio (raspbian 64bit)
* Pymodbus:
2.5.3rc1
* Modbus Hardware (if used): 

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async
client tcp / sync

### Description

What were you trying, what has happened, what went wrong, and what did you expect?

I am maintaining the modbus integration in Home Assistant, and I have lately received a number of issues that the reconnect does not work.

Some users tell it happens daily others that it happens from time to time. It just so happens that my production system (Raspberry PI 4 with hassio) started to have the same issue.

For some reason the solar inverter fails to respond:
```
Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```
That causes pymodbus to start reconnecting:
```
Connection to (solar.lan, 502) failed: timed out
```
But is cannot get a connect, because the primary socket is not closed properly (I traced open sockets)

We have a "special" command in home assistant that allows me to close pymodbus, and start it again, doing that solves the issue until next time.

### Code and Logs

```python
 # code and logs here.
https://github.com/home-assistant/core/blob/dev/homeassistant/components/modbus/modbus.py
```

Extract from the log:
```
2021-09-07 19:55:54 ERROR (SyncWorker_0) [homeassistant.components.modbus.modbus] Pymodbus: solar: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2021-09-07 20:17:27 ERROR (SyncWorker_0) [pymodbus.client.sync] Connection to (solar.lan, 502) failed: timed out
2021-09-07 20:17:30 ERROR (SyncWorker_1) [pymodbus.client.sync] Connection to (solar.lan, 502) failed: timed out
2021-09-07 20:17:33 ERROR (SyncWorker_2) [pymodbus.client.sync] Connection to (solar.lan, 502) failed: timed out
...
2021-09-08 16:47:43 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (solar.lan, 502) failed: timed out
reloaded pymodbus.
```

dhoomakethu, 2021-09-14:
There is a keyword option `reset_socket` which I believe is set to `False` by default on `2.5.3rc1` , did you try setting that to `True` to see if the socket gets reset on error?

janiversen, 2021-09-14:
I should have said it is a tcp connection.

Setting “reset_socket=True” does not change anything. I added the parameter on my production system, and first time the solar inverter had a problem pymodbus went into a loop trying to connect. A restart solved the problem until next time.

I am happy to help with more debugging etc.

dhoomakethu, 2021-09-14:
@janiversen Do you mean, the client tries to reconnect automatically ? Or is it the home assistant modbus client which tries to run transactions in loop ?

dhoomakethu, 2021-09-14:
And was this working before with earlier versions of pymodbus?

janiversen, 2021-09-14:
my software the client only calls read* in pymodbus every couple of seconds. It is pymodbus that tries to reconnect.

the “cannot reconnect” issues started a couple of months ago, could be around the time I updated to the release candidate. Let me see if I can try with 2.5.2

janiversen, 2021-09-14:
Nope inverter is down again. Anything I can do to help you narrow it down?

dhoomakethu, 2021-09-15:
@janiversen can you explicitly set `reset_socket=False` in the client initialisation and see if that helps?

janiversen, 2021-09-15:
That did not change a lot, but retry_on_empty=true seems to have solved my issue.

I do however think there is a problem hidden in all this. I will try to make test server and see if I can make something reproducible.

I use StartTcpServer with custom messages, but I have not found a way to add a function that allows modifications at the level of tcp send, any suggestions ?

dhoomakethu, 2021-09-15:
Have you tried Modbus server REPL ? It has features to control the response from the server.

```
(pymodbus) ➜  pymodbus git:(2.5.3) ✗ pymodbus.server --repl run --modbus-server tcp --modbus-framer socket --modbus-port 5020

__________                          .______.                    _________
\______   \___.__. _____   ____   __| _/\_ |__  __ __  ______  /   _____/ ______________  __ ___________
 |     ___<   |  |/     \ /  _ \ / __ |  | __ \|  |  \/  ___/  \_____  \_/ __ \_  __ \  \/ // __ \_  __ \
 |    |    \___  |  Y Y  (  <_> ) /_/ |  | \_\ \  |  /\___ \   /        \  ___/|  | \/\   /\  ___/|  | \/
 |____|    / ____|__|_|  /\____/\____ |  |___  /____//____  > /_______  /\___  >__|    \_/  \___  >__|
           \/          \/            \/      \/           \/          \/     \/                 \/

SERVER > help
Available commands:
clear                                        Clears screen
manipulator                                  Manipulate response from server.
Usage: manipulator response_type=|normal|error|delayed|empty|stray
	Additional parameters
		error_code=<int>
		delay_by=<in seconds>
		clear_after=<clear after n messages int>
		data_len=<length of stray data (int)>

	Example usage:
	1. Send error response 3 for 4 requests
	   manipulator response_type=error error_code=3 clear_after=4
	2. Delay outgoing response by 5 seconds indefinitely
	   manipulator response_type=delayed delay_by=5
	3. Send empty response
	   manipulator response_type=empty
	4. Send stray response of lenght 12 and revert to normal after 2 responses
	   manipulator response_type=stray data_len=11 clear_after=2
	5. To disable response manipulation
	   manipulator response_type=normal
SERVER >

```

dhoomakethu, 2021-09-30:
@janiversen any updates on this with explicti `reset_socket=False` ?

janiversen, 2021-09-30:
Ups sorry got held up. Yes but it is not good news, the problem persist, an error occurs, socket seems to stay open and pumodbus cannot reconnect.

I still have to try it out with REPL, will get some time this sunday.