pymodbus-dev/pymodbus #673: unit parameter

ferdymercury, 2021-10-08:
### Versions

GitHub master

### Description

When using pymodbus within an IDE like Spyder3, some ModbusSerialClient functions contain a 'hidden' parameter, namely one that does not appear as autocompletion, and that can only be found by looking in the docstrings by opening the source code. I am referring to parameter 'unit' in https://github.com/riptideio/pymodbus/blob/master/pymodbus/repl/client/mclient.py
[image]

It's only in the docstrings that one can find about how this parameter can be tuned. I needed to tune it from default's 0 to 1 and could not find how in the beginning.

[image]

I suggest to change the function declaration as follows:

`def read_holding_registers(self, address, count=1, unit=0, **kwargs):`

There are several more functions with that hidden parameter in that file.

iceisfun, 2022-01-22:
Should functions like read_holding_registers have a custom timeout parameter?

dhoomakethu, 2022-01-24:
@iceisfun the read time out is applicable for all transactions, Do you have a specific use case why you would need that ?

pkubanek, 2022-08-16:
@ferdymercury can you please check my fix really fix the issue? https://github.com/pkubanek/pymodbus. Thanks

janiversen, 2022-08-16:
Just had a quick view of your commit, looks good, suppose you are going to submit a PR at some point in time ?

ferdymercury, 2022-08-16:
Thanks for the fix! I tried to install and check, but my OS is a bit old and I get an error with the requirements:

```
ERROR: Could not find a version that satisfies the requirement bandit==1.7.4 (from versions: 0.10.0, 0.10.1, 0.11.0, 0.12.0, 0.13.0, 0.13.1, 0.13.2, 0.14.0, 0.14.1, 0.15.0, 0.15.1, 0.15.2, 0.16.0, 0.16.1, 0.16.2, 0.17.0, 0.17.2, 0.17.3, 1.0.1, 1.1.0, 1.2.0, 1.3.0, 1.4.0, 1.5.0, 1.5.1, 1.6.0, 1.6.1, 1.6.2, 1.6.3, 1.7.0, 1.7.1, 1.7.2)
ERROR: No matching distribution found for bandit==1.7.4
```

But otherwise it looks quite good!

janiversen, 2022-08-16:
bandit is only needed for development not for runtime.

janiversen, 2022-08-16:
I mean it is only needed for Pymodbus development, not if you develop an application.

pkubanek, 2022-08-17:
PR: https://github.com/riptideio/pymodbus/pull/1041

PR #1041 (2022-08-17): Add unit to arguments
Fix #673. Adds unit to argument list for selected functions.
commit 34acafd0 (2022-08-17): Add unit to arguments (#1041)

* Fix #673
closed 2022-08-17 by janiversen