pymodbus-dev/pymodbus #680: ModbusTcpClient.is_socket_open() always true

Jeroen6, 2021-11-04:
### Versions

* Python: 3.9.1
* OS: Windows 10.0.19043 Build 19043
* Pymodbus: pymodbus-2.5.3
* Modbus Hardware (if used):  wago 750-362

### Pymodbus Specific
* Client: tcp - sync

### Description

is_socket_open() keeps returning True when the connection is closed by something other than close().
For example, when the remote node issues a FIN due to 60 second idle timeout.

Plus trying to catch the ConnectionException(msg) that will be raised when trying to use the closed socket gives a NameError, but that is maybe my problem.

is_socket_open() should return false when the tcp connection is closed.

### Code and Logs

```python
from pymodbus.client.sync import ModbusTcpClient
import time

# this server times out connections after 60 seconds,
modbus = ModbusTcpClient("192.168.22.253")
result = modbus.read_coils(0)
print("read coil: {}".format(result.bits[0]))
print("sleep 61")
time.sleep(61)
print("is_socket_open {}".format(modbus.is_socket_open()))
print("read coil:")
modbus.read_coils(0)
```
```
read coil: True
sleep 61
is_socket_open True
read coil:
Traceback (most recent call last):
  File "[...]\is_socket_open.py", line 11, in <module>
    modbus.read_coils(0)
  File "[...]\pymodbus\client\common.py", line 48, in read_coils
    return self.execute(request)
  File "[...]\pymodbus\client\sync.py", line 109, in execute
    return self.transaction.execute(request)
  File "[...]\pymodbus\transaction.py", line 178, in execute
    broadcast=broadcast
  File "[...]\pymodbus\transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "[...]\pymodbus\transaction.py", line 326, in _recv
    read_min = self.client.framer.recvPacket(min_size)
  File "[...]\pymodbus\framer\__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "[...]\pymodbus\client\sync.py", line 89, in recv
    return self._recv(size)
  File "[...]\pymodbus\client\sync.py", line 298, in _recv
    size, data, time.time() - time_)
  File "[...]\pymodbus\client\sync.py", line 342, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.22.253:502): Connection unexpectedly closed 0.000000 seconds into read of 8 bytes without response from unit before it closed connection
```
```
212	1.488505	192.168.22.254	192.168.22.253	TCP	66	49729 → 502 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM=1
213	1.489174	192.168.22.253	192.168.22.254	TCP	66	502 → 49729 [SYN, ACK] Seq=0 Ack=1 Win=16000 Len=0 MSS=1460 SACK_PERM=1 WS=1
214	1.489218	192.168.22.254	192.168.22.253	TCP	54	49729 → 502 [ACK] Seq=1 Ack=1 Win=1051136 Len=0
216	1.508669	192.168.22.254	192.168.22.253	Modbus/TCP	66	   Query: Trans:     1; Unit:   0, Func:   1: Read Coils
217	1.509906	192.168.22.253	192.168.22.254	Modbus/TCP	64	Response: Trans:     1; Unit:   0, Func:   1: Read Coils
220	1.554718	192.168.22.254	192.168.22.253	TCP	54	49729 → 502 [ACK] Seq=13 Ack=11 Win=1051136 Len=0
6283	61.509557	192.168.22.253	192.168.22.254	TCP	60	502 → 49729 [FIN, ACK] Seq=11 Ack=13 Win=16000 Len=0
6284	61.509607	192.168.22.254	192.168.22.253	TCP	54	49729 → 502 [ACK] Seq=13 Ack=12 Win=1051136 Len=0
6424	62.522744	192.168.22.254	192.168.22.253	Modbus/TCP	66	   Query: Trans:     2; Unit:   0, Func:   1: Read Coils
6425	62.522842	192.168.22.254	192.168.22.253	TCP	54	49729 → 502 [FIN, ACK] Seq=25 Ack=12 Win=1051136 Len=0
6426	62.537341	192.168.22.253	192.168.22.254	TCP	60	502 → 49729 [ACK] Seq=12 Ack=14 Win=16000 Len=0
6438	62.853117	192.168.22.254	192.168.22.253	TCP	65	[TCP Retransmission] 49729 → 502 [FIN, PSH, ACK] Seq=14 Ack=12 Win=1051136 Len=11
6439	62.861278	192.168.22.253	192.168.22.254	TCP	60	[TCP Dup ACK 6426#1] 502 → 49729 [ACK] Seq=12 Ack=14 Win=16000 Len=0
6513	63.457088	192.168.22.254	192.168.22.253	TCP	65	[TCP Retransmission] 49729 → 502 [FIN, PSH, ACK] Seq=14 Ack=12 Win=1051136 Len=11
6514	63.471806	192.168.22.253	192.168.22.254	TCP	60	[TCP Dup ACK 6426#2] 502 → 49729 [ACK] Seq=12 Ack=14 Win=16000 Len=0
6729	64.667099	192.168.22.254	192.168.22.253	TCP	65	[TCP Retransmission] 49729 → 502 [FIN, PSH, ACK] Seq=14 Ack=12 Win=1051136 Len=11
6730	64.667427	192.168.22.253	192.168.22.254	TCP	60	502 → 49729 [RST] Seq=12 Win=0 Len=0
```

SvenBunge, 2021-11-25:
I have the same issue (and not able to update to 3.x because the device is only providing python 2.7):

Got the same error and it seems that the peer is closing the connection sometimes silently. It's not about a timeout, the peer is closing it between 8 reads in a row. Looking into the mentioned method:
```
    def is_socket_open(self):
        return True if self.socket is not None else False
```

This is only checking that the socket is set. It is unset by the `close()` method but from my perspective not if the peer just closes the connection.

In constants.py a Reconnect constant is set with 0. But it doesn't seems be used by any method - and not able to increase the reconnect value ;-) 

Long story short: Can we please create a reconnect ability in the API to handle closed sockets graceful? Retrying it N times and then throwing a ConnectionException.

janiversen, 2022-05-31:
Solved on current dev.

closed 2022-05-31 by janiversen