pymodbus-dev/pymodbus #682: Increase the usage of augmented assignment statements

elfring, 2021-11-12:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
(:point_right: Please check also for questionable change suggestions because of an evolving search pattern.)
```
lokal$ perl -p -i.orig -0777 -e 's/^(?<indentation>\s+)(?<target>\S+)\s*=\s*\k<target>[ \t]*(?<operator>[+\-%&|^@]|\*\*?|\/\/?|<<|>>)/$+{indentation}$+{target} $+{operator}=/gm' $(find ~/Projekte/pymodbus/lokal -name '*.py')
```

:crystal_ball: How will the development interests evolve further also according to update candidates in 24 lines of this software?

swamper123, 2021-11-15:
When we drop python2 (and perhaps everything below 3.7) support, we will gain more oppurtunities to improve the code (not just aufmented assinment statements alone). :)

elfring, 2021-11-15:
:thought_balloon: Do any factors hinder the wider application of [functionality which became generally available with Python 2](https://docs.python.org/3/whatsnew/2.0.html#augmented-assignment "Augmented assignments")?

elfring, 2021-12-18:
Would you like to improve the source code any further also for this software by using more augmented assignment statements? :thinking:

swamper123, 2021-12-20:
If we update to Python3 ( see e.g.  #564 ), I don't see why we shouldn't ... Perhaps not with the first bigger release, but built on top of that. A general cleanup after that, would be good.

elfring, 2021-12-20:
:thought_balloon: I imagine that the affected source code can be improved earlier according to the suggested transformation pattern.

swamper123, 2021-12-20:
If you got a tool to help or you got sth. in the wings, feel free to make a PR on that.
Every step of improvement is a step forwards.

elfring, 2021-12-20:
Would you dare to try another transformation approach out which would be similar to my Perl command example? :thinking:

swamper123, 2021-12-20:
I have no clue about Perl, so I wouldn't try using it. ^^ I would use other Python codestyle related tools (guess sth. like a PEP8 checker shall mention it).
In the end a Pythonfile is a Textfile. Also, if you pull the repo, run your script, and commit your stuff, you'll see what changes have been made and if tests break. With a PR, you can't do any harm to this project (until sb. merges it). ^^

elfring, 2021-12-20:
:thought_balloon: Would you like to become more familiar with the application of named capture groups (as they are used in the shown regular expressions for another source code search and replace approach)?

How do you think about to influence the evolution any further for more advanced software analysis tools (besides my feature requests)? :thinking:

swamper123, 2021-12-20:
If I would take some time on the regex topic again, I could handle it. ^^
Before that discussion is slipping to much into offtopic, I am a big fan of analysis tools, but haven't got the ressources at the moment to contribute something.

elfring, 2021-12-20:
Some programming languages support operators with a special property:
* augmented
* combined
* compound

Thus corresponding change possibilities can occasionally be reconsidered, can't they? :thinking:

janiversen, 2022-05-10:
Feel free to submit a PR.
Closing this since there are no problem nor a concrete solution of improvement.

closed 2022-05-10 by janiversen