pymodbus-dev/pymodbus #687: REPL console/server no connection

levmarki, 2021-11-29:
### Versions

* Python: 3.8.10
* OS: Ubuntu 20.04
* Pymodbus: last
* Modbus Hardware (if used): Fatek

### Pymodbus Specific
* Server: tcp/rtu/ascii - sync/async
* Client: tcp/rtu/ascii - sync/async

### Description
I'm trying to use Fatek PLC through TCP/IP Modbus.
As the first stage I tried to run REPL console and server on the same computer (I also tried to run on different PCs). The client.connect returns true, while read_coils returns http error.

What should I do?
Thank you all in advance

### Code and Logs

```python
SERVER > 2021-11-29 17:32:43,920 MainThread      ERROR    web_protocol   :368      Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/aiohttp/web_protocol.py", line 288, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
  File "aiohttp/_http_parser.pyx", line 546, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message="Bad status line 'invalid HTTP method'"

```

swamper123, 2021-11-29:
Could you provide a minimum setup, what you have done?

levmarki, 2021-12-02:
Hello,
yes, sure.
I opened two terminals, one with server and one with console.
When I wrote client.connect I got 'true'
When I wrote client.read_coils address... 
I got the error listed below.
The OS is Ubuntu 20.04.

closed 2022-01-07 by github-actions[bot]