pymodbus-dev/pymodbus #692: do support  the tcp-listener mode(modbus client base on tcp server)?

IndustrialCommunication, 2021-12-15:
commonly,modbus tcp client is tcp client mode, in some cloud case, need use tcp server mode listen the tcp client to connect itself.

dhoomakethu, 2021-12-16:
@IndustrialCommunication have you looked in to [isinstance(b, bytes) ](https://github.com/riptideio/pymodbus/blob/master/examples/contrib/remote_server_context.py) and https://github.com/riptideio/pymodbus/blob/master/examples/contrib/serial_forwarder.py ? This should be something you are looking for.

IndustrialCommunication, 2021-12-17:
that is different,  that is serial_forwarder,this is modbus client as a tcp server to listen the modbus slave to connect
[image]

closed 2022-01-22 by github-actions[bot]